platformio/platform-atmelmegaavr #1: Support for megaAVR 0-Series, Arduino Uno WiFi Rev2

sstaub, 2019-10-16:
Can you add support for the new megaAVR 0-series boards, like Arduino Wifi Rev. 2,
ATmega4809-XPRO (ATMEGA4809-XPRO) and ATmega4809 Curiosity Nano Evaluation kit.
The last costs under 10€ inclusive mEDBG https://www.microchip.com/DevelopmentTools/ProductDetails/DM320115
The Arduino core is released at https://github.com/arduino/ArduinoCore-megaavr
I think there is a new platform necessary.

ivankravets, 2019-10-16:
@valeros  added initial support for megAVR. Could you help with testing?

```
[env:myenv]
platform = https://github.com/platformio/platform-atmelmegaavr.git
framework = arduino
board = nona4809

; or
; board = uno2018
```

See supported boards in https://github.com/platformio/platform-atmelmegaavr/tree/develop/boards

Thanks!

sstaub, 2019-10-16:
Hi can upload a simple blinky with serial output, but nothing happens, no blink no output to console.

valeros, 2019-10-16:
@sstaub But it works with Arduino IDE, right? What board do you have?

sstaub, 2019-10-16:
I have different boards: 
Arduino UNO Wifi2 (Uno2018?)
Arduino Nano Every
Microchip 4809 Curiosity Nano.
UNO Wifi2 works on Arduino IDE (1.8.10) but not on PIO

valeros, 2019-10-16:
`Arduino Nano Every` doesn't work as well? Did you select any special settings in Arduino IDE?

sstaub, 2019-10-16:
What is the right board config for UNO WIFI 2 and Nano Every, maybe uno2018 is only a variant?

valeros, 2019-10-16:
I used the same names as in Arduino IDE:
`uno2018` - Arduino Uno WiFi Rev2
`nona4809` - Arduino Nano Every
PS: `nona4809` name looks like a typo (probably it should be `nano4809`), but let's be consistent with Arduino IDE for now.

sstaub, 2019-10-16:
Same on nona aka Nano Every, can upload but there is no code execution.

valeros, 2019-10-16:
What project are you trying to compile? Did you select any special settings in Arduino IDE?

sstaub, 2019-10-16:
How to do arduino-external-libs ?
There are no special setting in the Arduino IDE
Upload to UNO Wifi2 works with mEDGB and AVRISP mkII

`#include "Arduino.h"
#define LED LED_BUILTIN

// the setup function runs once when you press reset or power the board
void setup() {
	// initialize digital pin PB1 as an output.
	pinMode(LED, OUTPUT);
	Serial.begin(9600);
}

// the loop function runs over and over again forever
void loop() {
	Serial.print("time milliseconds ");
	Serial.println(millis());
	digitalWrite(LED, HIGH);   // turn the LED on (HIGH is the voltage level)
	delay(500);              // wait for a second
	digitalWrite(LED, LOW);    // turn the LED off by making the voltage LOW
	delay(500);              // wait for a second
}`

sstaub, 2019-10-16:
Eher usefull information about the boards and variants on https://github.com/MCUdude/MegaCoreX

valeros, 2019-10-16:
> There usefull information about the boards and variants on https://github.com/MCUdude/MegaCoreX

At the moment let's concentrate on the official core.

> Upload to UNO Wifi2 works with mEDGB and AVRISP mkII

Why do you need an external programmer? Could you please upload your project in Arduino IDE and attach here the entire log (compilation and upload). Thanks!

sstaub, 2019-10-16:
I don't use an external programmer, the UNO has a mEDGB on board, so with a future firmware it should possible to debug.
Also there is need an option if Register Emulation is used or not.

sstaub, 2019-10-16:
Der Sketch verwendet 2990 Bytes (6%) des Programmspeicherplatzes. Das Maximum sind 48640 Bytes.
Globale Variablen verwenden 175 Bytes (2%) des dynamischen Speichers, 5969 Bytes für lokale Variablen verbleiben. Das Maximum sind 6144 Bytes.
/Users/sstaub/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/Users/sstaub/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega4809 -cxplainedmini_updi -Pusb -b115200 -e -D -Uflash:w:/var/folders/3d/9y3knfj54cb1c6f_1c_887qw0000gn/T/arduino_build_792561/blinky.ino.hex:i -Ufuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x02:m -Uflash:w:/Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/bootloaders/atmega4809_uart_bl.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/sstaub/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf"
         User configuration file is "/Users/sstaub/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
         Overriding Baud Rate          : 115200
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega4809
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no         61   61      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9    0      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           usersig        0     0     0    0 no         64   64      0     0     0 0x00 0x00
           flash          0     0     0    0 no      49152  128      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        256   64      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Atmel AVR XplainedMini in UPDI mode
         ICE hardware version: 0
         ICE firmware version: 1.19 (rel. 57)
         Serial number   : 05B01296295C5F3F7E3F
         Vtarget         : 5.00 V
         JTAG clock megaAVR/program: 0 kHz
         JTAG clock megaAVR/debug:   0 kHz
         JTAG clock Xmega: 0 kHz
         PDI clock Xmega : 0 kHz

avrdude: Partial Family_ID returned: "mega"
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.04s

avrdude: Device signature = 0x1e9651 (probably m4809)
[... 56 lines cut ...]
avrdude: load data fuse8 data from input file 0x02:
avrdude: input file 0x02 contains 1 bytes
avrdude: reading on-chip fuse8 data:

Reading | ################################################## | 100% 0.02s

avrdude: verifying ...
avrdude: 1 bytes of fuse8 verified
avrdude: reading input file "/Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/bootloaders/atmega4809_uart_bl.hex"
avrdude: writing flash (280 bytes):

Writing | ################################################## | 100% 0.12s

avrdude: 280 bytes of flash written
avrdude: verifying flash memory against /Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/bootloaders/atmega4809_uart_bl.hex:
avrdude: load data flash data from input file /Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/bootloaders/atmega4809_uart_bl.hex:
avrdude: input file /Users/sstaub/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.5/bootloaders/atmega4809_uart_bl.hex contains 280 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.10s

avrdude: verifying ...
avrdude: 280 bytes of flash verified

avrdude done.  Thank you.

valeros, 2019-10-16:
Do you use this "Register Emulation"?

sstaub, 2019-10-16:
Personally No, but on Arduino IDE I think it is standard.
[image]

valeros, 2019-10-16:
OK, strange but I thought the emulation is default choice.  Try to add the following line to your `platformio.ini`
```
build_unflags = -DUNO_WIFI_REV2_328MODE
```

sstaub, 2019-10-16:
No changes, upload ok but no execution.

valeros, 2019-10-16:
Could you please try a special version of the platform:
```
platform = https://github.com/platformio/platform-atmelmegaavr.git#feature/dont-archive-framework-core
```

sstaub, 2019-10-16:
Great, it works.

sstaub, 2019-10-16:
also with Nano Every

ivankravets, 2019-11-18:
Finally! https://github.com/platformio/platform-atmelmegaavr/releases/tag/v1.0.0

closed 2019-11-18 by ivankravets