platformio/platform-atmelmegaavr #2: Support for  MegaCoreX

sstaub, 2019-11-19:
A support of the MegaCoreX will helpful because there are other boards with MegaAvr0 chips,
(Curiosity Nano, ATmega4809 Xplained Pro ...) which are not supported by Arduino directly.
https://github.com/MCUdude/MegaCoreX
Thank you for support of the original Arduino boards.

tombull, 2020-06-06:
I would love to see this!

MCUdude, 2020-06-30:
@valeros what do you need me to do in order make this happen? the current MegaCoreX release is ready to be used with PlatformIO. I can provide the necessary manifest files if you want.

However, there are a few things that need to be addressed.

* The official Arduino core for "megaavr" devices uses avr-gcc 7.3.0 with a set of patches. The reason for this was apparently because they had trouble getting 5.4.0 to work like it would. Maybe PlatformIO should use 7.3.0 as well? You'll probably have to look at the official Arduino toolchain to see what they have patched and what they have left out.
* The latest "[AT-pack](http://packs.download.atmel.com)" also has to be applied (currently 1.4.351). This is important because ATmega808 and 809 won't work properly without it.
* avrdude.conf has to be modified to support the most common UPDI programmers. These are the mEDBG chip (ATmega32u4 based), nEDBG chip (SAMD21 based), Atmel ICE (updi) and JTAG2UPDI.

If we're going to do this, I suggest we start by addressing these issues. When everything seems to work as it should, I could start working on a script that would calculate fuses based on values from platformio.ini. I can also make a script that will load the correct bootloader too, similar to the script `platform-atmelavr` has.

When this is finished, adding support for the megaavr ATtinys through [megaTinyCore](https://github.com/SpenceKonde/megaTinyCore) requested in https://github.com/platformio/platform-atmelavr/issues/83 would be simple.

These are the Avrdude programmers that need to be added:
```
programmer
  id    = "xplainedpro_updi";
  desc  = "Atmel AVR XplainedPro in UPDI mode";
  type  = "jtagice3_updi";
  connection_type = usb;
  usbpid = 0x2111;
;

programmer
  id    = "xplainedmini_updi";
  desc  = "Atmel AVR XplainedMini in UPDI mode";
  type  = "jtagice3_updi";
  connection_type = usb;
  usbpid = 0x2145;
;

programmer
  id    = "curiosity_updi";
  desc  = "Curiosity in UPDI mode";
  type  = "jtagice3_updi";
  connection_type = usb;
  usbpid = 0x2175;
;

programmer
  id    = "atmelice_updi";
  desc  = "Atmel-ICE (ARM/AVR) in UPDI mode";
  type  = "jtagice3_updi";
  connection_type = usb;
  usbpid = 0x2141;
;

programmer
  id    = "powerdebugger_updi";
  desc  = "Atmel PowerDebugger (ARM/AVR) in UPDI mode";
  type  = "jtagice3_updi";
  connection_type = usb;
  usbpid = 0x2144;
;

programmer
  id    = "jtag2updi";
  desc  = "JTAGv2 to UPDI bridge";
  type  = "jtagmkii_pdi";
  connection_type = serial;
  baudrate = 115200;
;

```

And these are the targets. Note that this contains _all_ current megaavr parts, including all ATtinys:

```
#------------------------------------------------------------
# AVR8X family common values
#------------------------------------------------------------
[... 926 lines cut ...]

#------------------------------------------------------------
# ATmega4809
#------------------------------------------------------------

part parent    ".avr8x_mega"
    id        = "m4809";
    desc      = "ATmega4809";
    signature = 0x1E 0x96 0x51;

    memory "flash"
        size      = 0xC000;
        offset    = 0x4000;
        page_size = 0x80;
        readsize  = 0x100;
    ;

    memory "eeprom"
        size      = 0x100;
        offset    = 0x1400;
        page_size = 0x40;
        readsize  = 0x100;
    ;
;
```

Gurkengewuerz, 2020-10-14:
@MCUdude @ivankravets Is there any update? I really want to use platformio for a atmega4808 but because of no support i am trying other ide's that talk with arduino directly...

ivankravets, 2020-10-14:
Sorry for the delay, we have just prioritized this issue and will add support soon.

/cc @valeros

Gurkengewuerz, 2020-10-14:
Thanks! Would be so great to see support soon. :)

MCUdude, 2020-10-14:
I'm ready to continue working on the MegaCoreX implementation whenever @ivankravets or @valeros is!

Gurkengewuerz, 2020-10-15:
@MCUdude would be cool if you three could come to an arrangement.

Question aside: Is there actually an official roadmap?

valeros, 2020-10-15:
Hi guys, sorry for the delay, let's try to sort this out. According to @MCUdude hints above I prepared:
- `GCC 7.3.0` with the patches from `Atmel ATmega Series Device Support (1.5.362)`
- Patched `avrdude` with new programmers and targets in `avrdude.conf`
- Package with the latest `MegaCoreX` v1.0.5

CI successfully built the examples, so we just need to agree on supported upload methods and flags.

MCUdude, 2020-10-15:
Awesome! I'm busy this weekend, but I'll get back to you on Monday. I do have a plan 😊

Gurkengewuerz, 2020-10-15:
Will it be possible to change the oscillator to external, set the bod, etc.? 👁️

MCUdude, 2020-10-16:
The first step would be to get it up and running. The next step would be to create two scripts. One to calculate the fuses, and one to load the correct bootloader. So @Gurkengewuerz, that is the plan. Similar functionality as with Minicore and MightyCore.

MCUdude, 2020-10-18:
> CI successfully built the examples, so we just need to agree on supported upload methods and flags.

@valeros the upload flags used at the moment can be found in the MegaCoreX platform.txt file. When it comes to the upload method, it's possible to upload using a dedicated programmer, or through a bootloader - Optiboot. This means the entire upload procedure will be very similar to what's already done with Mini/Mega/Major/MightyCore.

We should also discuss a fuses/bootloader script as well. As a start, I will try to implement ATmega808/1608/3208/4808 and ATmega809/1609/3209/4809, but there are no reason why the scripts can't support the entire tiny0, tiny1, and AVR DA/DB series later.

valeros, 2020-10-21:
Hi @MCUdude  I've pushed an initial implementation of fuses and bootloader programming for Arduino boards. Unfortunately I don't have any atmega4809-based board at hand, so it would be great if you could test new changes. But there is one thing that bothering me, in the current implementation. In Arduino core they program application, bootloader and set fuses in a single `avrdude` command. Can we safely split that single command into three independent phases?

MCUdude, 2020-10-21:
@valeros I'll test!

> But there is one thing that bothering me, in the current implementation. In Arduino core they program application, bootloader and set fuses in a single avrdude command. Can we safely split that single command into three independent phases?

Flashing, bootloadering and fuses can safely be split into three separate tasks. The only thing you have to be aware of is that the bootloader is located at the beginning of the flash memory, and not the end like on classic AVRs. This means that programs compiled to run on chips that are using a bootloader need a 512 byte (0x200) offset (configured the BOOTEND fuse). When no bootloader is in use, the program starts at address 0x0000.

https://github.com/MCUdude/MegaCoreX/blob/master/megaavr/boards.txt#L176

Here's a blink hex file where a bootloader is already installed. Note the 0x200 offset:

```ihex
:100200000C9448010C945A010C945A010C945A0114
:100210000C945A010C945A010C945A010C945A01F2
:100220000C945A010C945A010C945A010C945A01E2
:100230000C945A010C945A010C945A010C945A01D2
:100240000C945A010C945A010C945A010C945A01C2
:100250000C945A010C945A010C945A010C945A01B2
:100260000C945A010C94D1010C945A010C945A012B
:100270000C945A010C945A010C945A010C945A0192
:100280000C945A010C945A010C945A010C945A0182
:1002900011241FBECFEFCDBFDFE3DEBF28E2A0E019
:1002A000B8E201C01D92A430B207E1F70E94F90143
:1002B0000C948A020C9400019091010A9F7E909305
:1002C000010A9091600493FF0AC0811104C088E084
:1002D00080936604089588E08093650408959FB72D
:1002E000F894811107C080917304877F8093730411
:1002F0009FBF0895809173048860F8CF8FB7F894FA
:100300002091002830910128409102285091032823
:10031000E091AA0AF091AB0A9091A60A90FF08C05A
:100320002F5F3F4F4F4F5F4FE091AA0AF091AB0A0A
:100330008FBFA8EEB3E00E94700224E0F695E79527
:100340002A95E1F76E0F7F1F811D911D0895CF92B1
:10035000DF92EF92FF920E947E016B017C0120E40C
:10036000C20E22E4D21E2FE0E21EF11C80E4C81669
:1003700082E4D8068FE0E806F10438F40E947E019A
:10038000C616D706E806F906C8F30E947E016C156A
:100390007D058E059F05C8F3FF90EF90DF90CF900D
:1003A00008951F920F920FB60F9211248F939F936F
:1003B000AF93BF938091002890910128A0910228CB
:1003C000B09103280196A11DB11D80930028909340
:1003D0000128A0930228B093032881E08093A60A05
:1003E000BF91AF919F918F910F900FBE0F901F9073
:1003F000189588ED90E084BF90936100809102068B
:1004000086608093020680910006816080930006DA
:10041000809102068F7C8065809302061092E2052F
:1004200083E08093E40581E08093030A9EEF90933C
:10043000270A9093260A80E880932D0A80932B0A3E
:100440008093290A80932C0A80932A0A8093280A91
:100450002BE02093000A1092E50537E03093810AE3
:1004600090938C0A80938D0A25E02093800A309324
:10047000910A90939C0A80939D0A2093900A30934E
:10048000A10A9093AC0A8093AD0A2093A00A10921F
:10049000A10A8FE79EE38093AC0A9093AD0A809106
:1004A000A50A81608093A50A1092A00A8091A00AF3
:1004B00081608093A00A789488E080936104C0E012
:1004C000D0E081E00E945C010E94A70180E00E94D0
:1004D0005C010E94A7012097A1F30E940000F1CFC8
:1004E0000E947B02A59F900DB49F900DA49F800D4C
:1004F000911D11240895A29FB001B39FC001A39F35
:10050000700D811D1124911DB29F700D811D11244C
:08051000911D0895F894FFCF3E
:0400000300000200F7
:00000001FF
```

I will start working on dynamic fuse calculation once this PR is all figured out and working as it should. BTW how are the MegaCoreX bootloaders loaded? I tried to run `pio run -t bootloader`, but I got an error saying ` *** Do not know how to make File target booloader (/Users/hans/Documents/PlatformIO/Projects/MegaCoreX_test/booloader).  Stop. `

valeros, 2020-10-21:
I've updated the bootloader script, please pull the latest changes. 

> Flashing, bootloadering and fuses can safely be split into three separate tasks.

Thanks! Just in case, does the order of tasks matter?

> This means that programs compiled to run on chips that are using a bootloader need a 512 byte (0x200) offset (configured the BOOTEND fuse)

Users can control this using `build.text_section_start = 0x200` option.

MCUdude, 2020-10-21:
>> Flashing, bootloadering and fuses can safely be split into three separate tasks.

>Thanks! Just in case, does the order of tasks matter?

Well, the bootloader is optional, but the user program will only run correctly if the right fuses are set, just like on regular AVRs.

>> This means that programs compiled to run on chips that are using a bootloader need a 512 byte (0x200) offset (configured the BOOTEND fuse)

> Users can control this using build.text_section_start = 0x200 option.

OK. This should be handled automatically though. It's very easy to detect. Add the 0x200 offset if `board_hardware.uart != no_bootloader`.

valeros, 2020-10-22:
> OK. This should be handled automatically though. It's very easy to detect. Add the 0x200 offset if board_hardware.uart != no_bootloader.

There are also two other boards with the official Arduino core, but I will check what I can do.

MCUdude, 2020-10-22:
> There are also two other boards with the official Arduino core, but I will check what I can do.

The bootloader was developed after the Arduino Uno Wifi Rev2 and Arduino Nano Every was released. These two boards have an on-board UPDI programmer. The Uno Wifi Rev2 uses the 0x200 offset (no idea why, there are no bootloader, so the BOOTEND fuse should be 0x00), and the Nano every has 0x0000 offset

Gurkengewuerz, 2020-10-22:
If you guys need some hardware to test i have an Arduino Nano Every and in the next days ATmega4808/808 will be ordered from mouser. 👀

sstaub, 2020-10-22:
I can test with Arduino Nano Every, Uno WIFI 2 and Nano Curiosity board.

valeros, 2020-10-22:
@Gurkengewuerz It'd be great if someone could install the [upstream verion](https://docs.platformio.org/en/latest/platforms/atmelmegaavr.html#stable-and-upstream-versions) of the platform and try to program the `Arduino Nano Every` and `Uno WIFI 2` boards with a simple blink example.

sstaub, 2020-10-22:
How to differ between the official Arduino core and the MegaCoreX in platformio.ini ?

MCUdude, 2020-10-22:
I have a development board I can use to test ATmega808/1608/3208/4808. With and without bootloader.

Gurkengewuerz, 2020-10-22:
Somehow an Arduino Nao Every works.

### platformio.ini

```ini
[env:nano_every]
platform = https://github.com/platformio/platform-atmelmegaavr.git#develop
board = nano_every
framework = arduino
```

### Build Log

```
Processing nano_every (platform: https://github.com/platformio/platform-atmelmegaavr.git#develop; board: nano_every; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/nano_every.html
PLATFORM: Atmel megaAVR (1.1.1+sha.ebadbd9) > Arduino Nano Every
HARDWARE: ATMEGA4809 16MHz, 6KB RAM, 47.50KB Flash
PACKAGES:
 - framework-arduino-megaavr 1.8.6
 - tool-avrdude-megaavr 1.60300.191015 (6.3.0)
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Selected fuses:
[fuse2 = 0x01]
[fuse5 = 0xC9]
[fuse8 = 0x00]
Compiling .pio\build\nano_every\FrameworkArduinoVariant\variant.c.o
Compiling .pio\build\nano_every\FrameworkArduino\CDC.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\NANO_compat.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\UART.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\UART0.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\UART1.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\UART2.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\UART3.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\UNO_compat.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\USBCore.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\abi.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\api\Common.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\api\IPAddress.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\api\PluggableUSB.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\api\Print.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\api\Stream.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\api\String.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\hooks.c.o
Compiling .pio\build\nano_every\FrameworkArduino\main.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\new.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\wiring.c.o
Compiling .pio\build\nano_every\FrameworkArduino\wiring_analog.c.o
Compiling .pio\build\nano_every\FrameworkArduino\wiring_digital.c.o
Compiling .pio\build\nano_every\FrameworkArduino\wiring_pulse.S.o
Compiling .pio\build\nano_every\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\nano_every\FrameworkArduino\wiring_shift.c.o
Compiling .pio\build\nano_every\src\main.cpp.o
Linking .pio\build\nano_every\firmware.elf
Checking size .pio\build\nano_every\firmware.elf
Building .pio\build\nano_every\firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   7.0% (used 429 bytes from 6144 bytes)
Flash: [=         ]   8.4% (used 4072 bytes from 48640 bytes)
Configuring upload protocol...
AVAILABLE: jtag2updi
CURRENT: upload_protocol = jtag2updi
Looking for upload port...
Auto-detected: COM3
Forcing reset using 1200bps open/close on port COM3
Uploading .pio\build\nano_every\firmware.hex

avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.27s

avrdude: Device signature = 0x1e9651 (probably m4809)
avrdude: erasing chip
avrdude: reading input file ".pio\build\nano_every\firmware.hex"
avrdude: writing flash (4072 bytes):

Writing | ################################################## | 100% 2.99s

avrdude: 4072 bytes of flash written

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)

avrdude done.  Thank you.

Setting fuses
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
avrdude: jtagmkII_getsync(): sign-on command: status -1
```

@valeros strangely the program get flashed correctly BUT the jtagmkII_getsync get no timeout. its just posting `avrdude: jtagmkII_getsync(): sign-on command: status -1` until i press `STRG+C`. After that i got

```
^Cscons: *** [upload] Build interrupted.
Error: Aborted by user
The terminal process "C:\Users\nikla\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```

let me hear if it helps you. If you want me to test something else, just mention me.

MCUdude, 2020-10-22:
@Gurkengewuerz you're not building Nano Every with MegaCoreX, you're using the official core which lacks several hardware features that MegaCoreX offer.

>
> How to differ between the official Arduino core and the MegaCoreX in platformio.ini ?

I'm not sure it's implemented yet, but MegaCoreX have several pinouts to choose from:  
`48pin-standard`, `40pin-standard` `nano-every` and `uno-wifi`.

Try using this configuration:

```ini
[env:ATmega4808]
platform = atmelmegaavr
framework = arduino
board = ATmega4809
board_build.variant = nano-every
```

Gurkengewuerz, 2020-10-22:
@MCUdude oh good to know! So the following config should be fine?

```ini
[env:nano_every]
platform = https://github.com/platformio/platform-atmelmegaavr.git#develop
framework = arduino
board = ATmega4809
board_build.variant = nano-every
```

PlatformIO also updated to toolchain-atmelavr @ 2.70300.201015 now

valeros, 2020-10-22:
`MegaCoreX` should also be set as the core:

```ini
[env:nano_every]
platform = https://github.com/platformio/platform-atmelmegaavr.git#develop
framework = arduino
board = ATmega4809
board_build.core= MegaCoreX
board_build.variant = nano-every
```

Gurkengewuerz, 2020-10-22:
Using the config above the flash was successful!

### Build Log

```
Processing nano_every (platform: https://github.com/platformio/platform-atmelmegaavr.git#develop; framework: arduino; board: ATmega4809)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/ATmega4809.html
PLATFORM: Atmel megaAVR (1.1.1+sha.ebadbd9) > ATmega4809
HARDWARE: ATMEGA4809 16MHz, 6KB RAM, 48KB Flash
PACKAGES:
 - framework-arduino-megaavr-megacorex 1.0.5
 - tool-avrdude-megaavr 2.60300.201015 (6.3.0)
 - toolchain-atmelavr 2.70300.201015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 8 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\nano_every\FrameworkArduino\CDC.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\UART.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\UART0.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\UART1.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\UART2.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\UART3.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\UNO_compat.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\USBCore.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\WInterrupts.c.o
Compiling .pio\build\nano_every\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\abi.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\api\Common.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\api\IPAddress.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\api\PluggableUSB.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\api\Print.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\api\RingBuffer.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\api\Stream.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\api\String.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\hooks.c.o
Compiling .pio\build\nano_every\FrameworkArduino\main.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\new.cpp.o
Compiling .pio\build\nano_every\FrameworkArduino\wiring.c.o
Compiling .pio\build\nano_every\FrameworkArduino\wiring_analog.c.o
Compiling .pio\build\nano_every\FrameworkArduino\wiring_digital.c.o
Compiling .pio\build\nano_every\FrameworkArduino\wiring_pulse.S.o
Compiling .pio\build\nano_every\FrameworkArduino\wiring_pulse.c.o
Compiling .pio\build\nano_every\FrameworkArduino\wiring_shift.c.o
Compiling .pio\build\nano_every\src\main.cpp.o
Linking .pio\build\nano_every\firmware.elf
Checking size .pio\build\nano_every\firmware.elf
Building .pio\build\nano_every\firmware.hex
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  11.9% (used 731 bytes from 6144 bytes)
Flash: [=         ]   9.3% (used 4576 bytes from 49152 bytes)
Configuring upload protocol...
AVAILABLE: jtag2updi
CURRENT: upload_protocol = jtag2updi
Looking for upload port...
Auto-detected: COM3
Forcing reset using 1200bps open/close on port COM3
Waiting for the new upload port...
Uploading .pio\build\nano_every\firmware.hex

avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.27s

avrdude: Device signature = 0x1e9651 (probably m4809)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file ".pio\build\nano_every\firmware.hex"
avrdude: writing flash (4576 bytes):

Writing | ################################################## | 100% 3.37s

avrdude: 4576 bytes of flash written
avrdude: verifying flash memory against .pio\build\nano_every\firmware.hex:
avrdude: load data flash data from input file .pio\build\nano_every\firmware.hex:
avrdude: input file .pio\build\nano_every\firmware.hex contains 4576 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.67s

avrdude: verifying ...
avrdude: 4576 bytes of flash verified

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)

avrdude done.  Thank you.

================================================================================================= [SUCCESS] Took 12.54 seconds =================================================================================================
```

MCUdude, 2020-10-22:
@Gurkengewuerz ah, yes, the platform field has to be the github URL.

@valeros I think there is something weird going on with the build process. When I build a blank sketch in Arduino IDE, it compiles down to 476 bytes. However, with PlatformIO it compiles to 4340 bytes.

Gurkengewuerz, 2020-10-22:
@MCUdude just a side note: Maybe you want to edit you larger comments in this issue with a spoiler tag. Scrolling on my phone is a pain in the ass 🤣

````markdown
### Titel

</details>

```
Lorem ipsum dolor sit amet, consetetur sadipscing elitr, sed diam nonumy eirmod tempor invidunt ut labore et dolore magna aliquyam erat, sed diam voluptua. At vero eos et accusam et justo duo dolores et ea rebum. Stet clita kasd gubergren, no sea takimata sanctus est Lorem ipsum dolor sit amet. Lorem ipsum do
```

</details>

```

Gurkengewuerz, 2020-10-22:
@valeros @MCUdude oh true. can cornfirm that the compiled blink sketch in Arduino IDE is much smaller using the MegaCoreX. Could it be because of the `UARTX.cpp`, `wiring_*.cpp` etc. compiled with it in the PlatformIO implementation (see my build log)?

MCUdude, 2020-10-22:
> @valeros @MCUdude oh true. can cornfirm that the compiled blink sketch in Arduino IDE is much smaller using the MegaCoreX. Could it be because of the UARTX.cpp, wiring_*.cpp etc. compiled with it in the PlatformIO implementation (see my build log)?

I'm not sure what causes this. RAM usage is also way too high. I can confirm that this is also an issue when compiling for the "official" Nano Every and Uno Wifi Rev2.

sstaub, 2020-10-22:
Tried with UNO WIFI 2, can compile but upload fails:
I'm on macOS
```
Processing ATmega4809 (platform: https://github.com/platformio/platform-atmelmegaavr.git#develop; framework: arduino; board: ATmega4809)
----------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/ATmega4809.html
PLATFORM: Atmel megaAVR (1.1.1+sha.ebadbd9) > ATmega4809
HARDWARE: ATMEGA4809 16MHz, 6KB RAM, 48KB Flash
PACKAGES: 
 - framework-arduino-megaavr-megacorex 1.0.5 
 - tool-avrdude-megaavr 2.60300.201015 (6.3.0) 
 - toolchain-atmelavr 2.70300.201015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 9 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/ATmega4809/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  11.9% (used 731 bytes from 6144 bytes)
Flash: [==        ]  15.4% (used 7576 bytes from 49152 bytes)
Configuring upload protocol...
AVAILABLE: jtag2updi
CURRENT: upload_protocol = jtag2updi
Looking for upload port...
Auto-detected: /dev/cu.usbmodem14302
Forcing reset using 1200bps open/close on port /dev/cu.usbmodem14302
Waiting for the new upload port...
Uploading .pio/build/ATmega4809/firmware.hex
avrdude: jtagmkII_recv_frame(): timeout
avrdude: jtagmkII_getsync(): sign-on command: status -1
```
following ini:
```
[env:ATmega4809]
#platform = atmelmegaavr
platform = https://github.com/platformio/platform-atmelmegaavr.git#develop
framework = arduino
board = ATmega4809
board_build.variant = uno-wifi

build_flags = -Wl,-u,vfprintf -lprintf_flt -lm
```

Code:
```cpp
#include "Arduino.h"
#define LED LED_BUILTIN
// #define LED PIN_PF5 // Curiosity Nano ATmega 4809 MegaCoreX
// #define LED PIN_PC6 // Curiosity Nano AVR128DA48 DxCore

void setup() {
	pinMode(LED, OUTPUT);
	Serial.begin(9600);
	// Serial3.begin(9600); // Curiosity Nano ATmega 4809 MegaCoreX
	// Serial1.begin(9600); // Curiosity Nano AVR128DA48 DxCore
}

void loop() {
	Serial.print("time milliseconds ");
	Serial.println(millis());
	// Serial3.printf("millis %d\n", millis()); // Curiosity Nano ATmega 4809 MegaCoreX
	// Serial1.printf("Millis %d\n", millis()); // Curiosity Nano AVR128DA48 DxCore
	digitalWrite(LED, HIGH);
	delay(200);
	digitalWrite(LED, LOW);
	delay(200);
	}
```

sstaub, 2020-10-22:
Arduino Nano Every works, but UNO WIFI 2 not. How to upload to Nano Curiosity board which uses nEDBG?

valeros, 2020-10-22:
Thanks, the footprint issue should be fixed in the dev branch. Please update to the latest revision.

@sstaub Does the uploading via `jtag2updi` work in Arduino IDE with `UNO WIFI 2` and the `MegaCoreX` core? Could you please attach here the entire log from Arduino IDE?

MCUdude, 2020-10-22:
The Nano every does use `jtag2updi`. The Uno Wifi Rev2 uses `xplainedmini_updi`, and curiosity nano uses `curiosity_updi`.

valeros, 2020-10-22:
@sstaub Probably you need to change the upload protocol when programming the `Uno WiFi` board even if a variant from MegaCoreX is used.

sstaub, 2020-10-22:
Changed the upload protocol but failed also: (macOS 10.15.7)
[image]

sstaub, 2020-10-22:
Update:
same problem with Curiosity Nano

valeros, 2020-10-22:
`jtag2updi` is the default protocol and it requires upload port. If you want to use `xplainedmini_updi` you need to disable the `wait_for_upload_port` option.
```ini
[env:ATmega4809]
platform = https://github.com/platformio/platform-atmelmegaavr.git#develop
framework = arduino
board = ATmega4809

board_build.variant = uno-wifi
board_upload.require_upload_port = False
```

sstaub, 2020-10-22:
OK, this works for the UNO WIFI 2. Only the Curiosity board is missing.

sstaub, 2020-10-22:
OK, this works also for Curiosity nano. Thank you very much, hope we can do the same for the DxCore.

sstaub, 2020-10-22:
I made three new board definitions:
- curiosity_nano_4809
- uno_wifi_x
- nano_every_x
this allows to use Arduino and Microchip boards without extra definitions for MegaCoreX
If ok, can you add it please? I attached the definitions within the boards.zip
All variants are tested.
[boards.zip](https://github.com/platformio/platform-atmelmegaavr/files/5425787/boards.zip)

MCUdude, 2020-10-23:
> I made three new board definitions:
curiosity_nano_4809
uno_wifi_x
nano_every_x
this allows to use Arduino and Microchip boards without extra definitions for MegaCoreX
If ok, can you add it please? I attached the definitions within the boards.zip
All variants are tested.

The whole idea with the MegaCoreX implementation is that it's supposed to be generic. Nano Every and Uno Wifi Rev2 are just pinout variants that are supported. it would be much better if we could figure out a way to get rid of the `board_upload.require_upload_port` flag. AFAIK the only available UPDI programmer out there that *require* a serial power is jtag2updi. @valeros can't we just have some logic in the megaavr main.py that checks what programmer is currently being used, and tweak `board_upload.require_upload_port` automatically?

sstaub, 2020-10-23:
For a concrete board there are more things to handle than `board_upload.require_upload_port` in the .json board definition. e.g. for the curiosity board also the debugger capability should handled in future.

sstaub, 2020-10-25:
Added two other boards 
- atmega4809_xplained
- avi-iot_wr

both can compile but no tests of real boards
I updated some URLs
[boards.zip](https://github.com/platformio/platform-atmelmegaavr/files/5434340/boards.zip)

valeros, 2020-10-30:
@MCUdude I updated the build script, so `jtag2udpi` is now configured dynamically. If you need more time for preparing your PR with dynamic fuses, we can simply publish a new release with basic functionality.

MCUdude, 2020-10-30:
> @MCUdude I updated the build script, so jtag2udpi is now configured dynamically. If you need more time for preparing your PR with dynamic fuses, we can simply publish a new release with basic functionality.

Thanks! It will take me some time to get everything just the way I like it, so please release it as it is now, and prepare a new release when the dynamic fuses/bootloader functionality is in. But it would be great if you could just wait a little more before doing the release so I have time to test all chips and pinout variants for all targets. I should be done testing everything tonight

valeros, 2020-10-30:
@MCUdude No problem, thanks a lot for testing!

MCUdude, 2020-10-30:
I've found one issue so far.

The device signature for ATmega3208 and ATmega3209 in avrdude.conf is incorrect.

Current:
```
#------------------------------------------------------------
# ATmega3208
#------------------------------------------------------------

part parent    ".avr8x_mega"
    id        = "m3208";
    desc      = "ATmega3208";
    signature = 0x1E 0x95 0x52;

    memory "flash"
        size      = 0x8000;
        offset    = 0x4000;
        page_size = 0x80;
        readsize  = 0x100;
    ;

    memory "eeprom"
        size      = 0x100;
        offset    = 0x1400;
        page_size = 0x40;
        readsize  = 0x100;
    ;
;

#------------------------------------------------------------
# ATmega3209
#------------------------------------------------------------

part parent    ".avr8x_mega"
    id        = "m3209";
    desc      = "ATmega3209";
    signature = 0x1E 0x95 0x53;

    memory "flash"
        size      = 0x8000;
        offset    = 0x4000;
        page_size = 0x80;
        readsize  = 0x100;
    ;

    memory "eeprom"
        size      = 0x100;
        offset    = 0x1400;
        page_size = 0x40;
        readsize  = 0x100;
    ;
;
```

is supposed to be:

```
#------------------------------------------------------------
# ATmega3208
#------------------------------------------------------------

part parent    ".avr8x_mega"
    id        = "m3208";
    desc      = "ATmega3208";
    signature = 0x1E 0x95 0x30;

    memory "flash"
        size      = 0x8000;
        offset    = 0x4000;
        page_size = 0x80;
        readsize  = 0x100;
    ;

    memory "eeprom"
        size      = 0x100;
        offset    = 0x1400;
        page_size = 0x40;
        readsize  = 0x100;
    ;
;

#------------------------------------------------------------
# ATmega3209
#------------------------------------------------------------

part parent    ".avr8x_mega"
    id        = "m3209";
    desc      = "ATmega3209";
    signature = 0x1E 0x95 0x31;

    memory "flash"
        size      = 0x8000;
        offset    = 0x4000;
        page_size = 0x80;
        readsize  = 0x100;
    ;

    memory "eeprom"
        size      = 0x100;
        offset    = 0x1400;
        page_size = 0x40;
        readsize  = 0x100;
    ;
;
```

MCUdude, 2020-10-30:
OK, I couldn't wait until tonight. I've now tested all targets (ATmega4809/3209/1609/809 and ATmega4808/3208/1608/808) and all pinouts, and the only issue I've found is the device signature typo.

valeros, 2020-10-30:
@MCUdude Thanks, fixed package are in the registry.

MCUdude, 2020-10-30:
> @MCUdude Thanks, fixed packages are in the registry.

Perfect! ATmega3208/3209 works now 🙂

MCUdude, 2020-10-30:
FYI I did test a Curiosity nano ATmega4809 board with the on-board UPDI programmer. Works a treat!

I think this issue can be closed now.

closed 2020-10-30 by valeros