platformio/platform-atmelmegaavr #6: Unable to upload to Nano_Every

chrisinabox, 2020-09-29:
I have been having persistent issues uploading to the Nano_Every from PIO with the atmelmegaavr platform (sample verbose at bottom of issues).

It seems to be that the upload hangs as it forces a board reset and then waits for the upload port which I don't think is needed (I am no expert here). I have not had these issues with the Arduino IDE which is using the same version of avrdude (at this moment in time on my machine) so am fairly certain it's due to how PIO is handling uploads to this board. 

I have managed to successfully program the board by modifying the nano_every.json so these two lines are false:
`"require_upload_port": false,`
`"wait_for_upload_port": false`

Verbose output for non modified json is consistently as follows: 

```
<lambda>(["upload"], [".pio\build\nano_every\firmware.hex"])
AVAILABLE: jtag2updi
CURRENT: upload_protocol = jtag2updi
BeforeUpload(["upload"], [".pio\build\nano_every\firmware.hex"])
Auto-detected: COM23
Forcing reset using 1200bps open/close on port COM23
Waiting for the new upload port...
avrdude -v -p atmega4809 -C C:\Users\nocma4\.platformio\packages\tool-avrdude-megaavr\avrdude.conf -c jtag2updi -D -V -b 115200 -e -P "COM23" -U flash:w:.pio\build\nano_every\firmware.hex:i -Ufuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x00:m

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\nocma4\.platformio\packages\tool-avrdude-megaavr\avrdude.conf"

         Using Port                    : COM23
         Using Programmer              : jtag2updi
         Overriding Baud Rate          : 115200
avrdude: ser_open(): can't open device "\\.\COM23": Access is denied.

avrdude done.  Thank you.

*** [upload] Error 1
```

chrisinabox, 2020-09-29:
My mistake - only `"wait_for_upload_port": false`. The other line causes it to fail for somewhat obvious reasons.

closed 2020-10-15 by valeros

gurghet, 2020-11-24:
I'm having a similar issue. Is there any workaround? Following is my verbose shell.
```
PS C:\Users\gurgh\OneDrive\Documents\PlatformIO\Projects\Audiuino> pio run -t upload -v
Processing nano_every (platform: atmelmegaavr; board: nano_every; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------

CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/nano_every.html
PLATFORM: Atmel megaAVR (1.3.0) > Arduino Nano Every
HARDWARE: ATMEGA4809 16MHz, 6KB RAM, 47.50KB Flash
PACKAGES:
 - framework-arduino-megaavr 1.8.6
 - tool-avrdude-megaavr 1.60300.191015 (6.3.0) 
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Auto-detected: COM8
MethodWrapper(["checkprogsize"], [".pio\build\nano_every\firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.4% (used 26 bytes from 6144 bytes)
Flash: [          ]   3.0% (used 1456 bytes from 48640 bytes)
.pio\build\nano_every\firmware.elf  :

section                     size      addr

.data                          6   8398848

.text                       1346         0

.rodata                      104     17730

.bss                          20   8398854

.comment                      17         0

.note.gnu.avr.deviceinfo      64         0

.debug_aranges               192         0

.debug_info                10525         0

.debug_abbrev               8992         0

.debug_line                 1303         0

.debug_str                  3979         0

Total                      26548
<lambda>(["upload"], [".pio\build\nano_every\firmware.hex"])
AVAILABLE: jtag2updi
CURRENT: upload_protocol = jtag2updi
BeforeUpload(["upload"], [".pio\build\nano_every\firmware.hex"])
Use manually specified: COM8
Forcing reset using 1200bps open/close on port COM8
Waiting for the new upload port...
```
After a few minutes it will print
```
avrdude -v -p atmega4809 -C "C:\Users\gurgh\.platformio\packages\tool-avrdude-megaavr\avrdude.conf" -c jtag2updi -Ufuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x00:m -Ulock:w:0xc5:m -b 115200 -V -e -D -P "COM8" -U flash:w:.pio\build\nano_every\firmware.hex:i

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\gurgh\.platformio\packages\tool-avrdude-megaavr\avrdude.conf"

         Using Port                    : COM8
         Using Programmer              : jtag2updi
         Overriding Baud Rate          : 115200
```

commit 9ceb1838 (2020-10-15): Don't wait for upload port after reset for Nano Every // Resolve #6