platformio/platform-atmelmegaavr #7: Use pyupdi for UPDI uploads

MCUdude, 2020-10-30:
I'll just leave this here.

There is another way of uploading to UPDI-compatible AVR microcontrollers that don't include Avrdude and is open source. It's called `pyupdi`, and it is developed by Microchip themselves. The actual hardware is dead simple.
https://github.com/mraardvark/pyupdi

```
 From the pyupdi repo

                        Vcc                     Vcc
                        +-+                     +-+
                         |                       |
 +---------------------+ |                       | +--------------------+
 | Serial port         +-+                       +-+  AVR device        |
 |                     |      +----------+         |                    |
 |                  TX +------+   4k7    +---------+ UPDI               |
 |                     |      +----------+    |    |                    |
 |                     |                      |    |                    |
 |                  RX +----------------------+    |                    |
 |                     |                           |                    |
 |                     +--+                     +--+                    |
 +---------------------+  |                     |  +--------------------+
                         +-+                   +-+
                         GND                   GND

```

As the name implies, it's Python based, and PlatformIO is Python's best friend. Would it be possible to add support for pyupdi, if so as something one could call in `upload_command`? I can help out with testing!

```ini
[env:Upload_using_pyupdi]
upload_protocol = custom
upload_flags =
    -d 
    $BOARD_MCU 
    -C 
    $UPLOAD_PORT 
    -b 
    $UPLOAD_SPEED
upload_command = pyupdi.py $UPLOAD_FLAGS -f $SOURCE
```

valeros, 2020-10-30:
It seems that their device names a bit differ from ours, so your probably won't be able to use the `$BOARD_MCU` variable. Anyway, just install the pyudpi package via pip `pip install https://github.com/mraardvark/pyupdi/archive/master.zip` and it should work with this your slightly modified configuration:
```ini
...
upload_speed = 115200
upload_flags =
    -d
    mega3209
    -c
    $UPLOAD_PORT
    -b
    $UPLOAD_SPEED
upload_command = pyupdi $UPLOAD_FLAGS -f $SOURCE
```

MCUdude, 2020-10-30:
Cool, I'll give it a try a little later. So it wouldn't make sense to include it in the atmelmegaavr toolchain? It's easy to install pyupdi on a Mac/Linux computer, but on Windows, you'll probably have to install the whole python/pip package first, so the bar is a little higher.

Could pyupdi be installed automatically by PlatformIO if called?

valeros, 2020-10-30:
If someone uses PlatformIO they already have Python and pip installed, so you can type directly in the IDE terminal `python -m pip install https://github.com/mraardvark/pyupdi/archive/master.zip`

MCUdude, 2020-11-05:
It would be great if you could document this in your docs as an option for a custom upload command. I will also document it in the MegaCoreX repo

valeros, 2020-11-05:
Thanks, added!

closed 2020-11-05 by valeros