platformio/platform-atmelmegaavr #8: debug support for microchip development boards

sstaub, 2020-10-30:
The official development boards from microchip like the Curiosity Nano 4809 have a build in debug adapter.
Hope you find a way to use it.

maxgerhardt, 2021-02-13:
+1. 

A fork of avarice at https://github.com/Florin-Popescu/avarice can reportedly open a gdbserver using the EDGB programmer / debugger via the UPDI interface. But not some other devices of a different class, [see issue](https://github.com/Florin-Popescu/avarice/issues/2). Would love to see avarice development continue and being integrated here.

paulskirk53, 2021-02-14:
Hi, It would be great to see support for debugging on chips such as megaavr4809. My background is programming in C# to write drivers for astronomy equipment (using the ASCOM standard) and programming Arduino mega2560 boards to control robotics stuff. I am retired and if there's anything I can do to help (e.g. testing) I am happy to commit time on a regular basis. I hope this helps. I am new to Platformio, but just to observe that continual enhancement of the platform to include support for new boards etc is a key to engagement with PIO. I know it all takes a lot of time and commitment - thanks to all involved.
Paul

mraardvark, 2021-02-25:
Basic debugging is possible with:

https://github.com/stemnic/pyAVRdbg

(I hope to take this further when I get time... :/ )

maxgerhardt, 2021-02-25:
For reference, see also issue https://github.com/stemnic/pyAVRdbg/issues/8 -- would be nice to test this with an Atmel-ICE device, and maybe someday the final frontier, getting it to work with the built-in programming bridge chips on some boards like the Arduino Nano Every, might be achieved.

maxgerhardt, 2021-03-16:
FYI: Using an Arduino Pro Micro (ATMega32U4) loaded with the https://github.com/MCUdude/microUPDI firmware and https://github.com/stemnic/pyAVRdbg as the GDB server we have made progress to get a working debugger in PlatformIO, [see topic](https://community.platformio.org/t/debug-an-avr-4809-with-microupdi/19926/39?u=maxgerhardt) :) 

[image]

Some things are still missing though -- you can't load the firmware over the GDB server and I also don't yet know how to reset the target chip. I've opened [issues](https://github.com/stemnic/pyAVRdbg/issues) for that..

mraardvark, 2021-03-16:
Nice work :)
Out of interest - did you try it with your Atmel-ICE as well?

maxgerhardt, 2021-03-16:
I don't have such an expensive debug probe sadly 😅

mraardvark, 2021-03-16:
Ah, sorry - I thought you mentioned one earlier.  A fictitious one then :)
Reason I was asking is that the mEDBG (mega32U4) is a horrible UPDI bit-banger, and I would like to know how it performs.
Sawing the nEDBG off a Curiosity Nano would yield a nicer UPDI driver...

MCUdude, 2022-01-25:
I have absolutely no experience with avarice or setting up an environment capable of debugging an AVR, and the last time I _had_ to use a hardware debugger to figure out what was going on, I loaded the elf file from PlatformIO into Atmel Studio, which to my surprise actually worked! However, it would be really nice to have AVR hardware debugging working together with PlatformIO.

It looks like the AVaRICE project has been migrated over to Github: https://github.com/avrdudes/avarice

There's a branch where there has been some recent activity too: https://github.com/avrdudes/avarice/tree/jputcu_dev

I haven't read all the forums posts regarding this topic, so what more needs to be done before avarice+avr-gdb works with VSCode+PlatformIO (as long as the platformio.ini file is set up correctly)? I'd love to help out if possible, I have access to most AVR debuggers, so I can at least help out with testing.

N0cynym, 2022-06-10:
so is this basically a updi protocol issue or much more which needs tbd?