platformio/platform-atmelmegaavr #9: MegaCoreX dynamic fuses and bootloader

MCUdude, 2020-10-30:
Hi!
After you @valeros released the latest `atmelmegaavr` package I started working on the fuses/bootloader script, and it wasn't that much work really. As mentioned earlier, I am **NOT** a (professional) Python programmer. The scrips work, but they are probably terrible because I'm happy as long as they just do what I want.

First, here is the fuses script I've come up with:

<details>

```py
import sys
import os

from SCons.Script import ARGUMENTS, COMMAND_LINE_TARGETS, Import, Return

Import("env")

def get_WDTCFG_fuse():
    return 0x00

def get_BODCFG_fuse(bod):
    if bod == "4.3v":
        return 0xF4
    elif bod == "2.6v":
        return 0x54
    elif bod == "1.8v":
        return 0x14
    else: # bod disabled
        return 0x00

def get_OSCCFG_fuse(f_cpu, oscillator):
    if (f_cpu == "20000000L" or f_cpu == "10000000L" or f_cpu == "5000000L") and oscillator == "internal":
        return 0x02
    else:
        return 0x01

def get_TCD0CFG_fuse():
    return 0x00

def get_SYSCFG0_fuse(eesave, rstpin, uart):
    eesave_bit = 1 if eesave == "yes" else 0
    if rstpin == "gpio":
        if uart == "no_bootloader":
            rstpin_bit = 0
        else:
            rstpin_bit = 1
    else:
        rstpin_bit = 1
    return 0xC0 | rstpin_bit << 3 | eesave_bit

def get_SYSCFG1_fuse():
    return 0x06

def get_APPEND_fuse():
    return 0x00

def get_BOOTEND_fuse(uart):
    if uart == "no_bootloader":
        return 0x00
    else:
        return 0x02

def get_LOCKBIT_fuse():
    return 0xC5

def print_fuses_info(fuse_values, fuse_names, lock_fuse):
    print("\nSelected fuses:")
    print("------------------------")
    for idx, value in enumerate(fuse_values):
        if value:
            print("[fuse%d / %s = 0x%s]" % (idx, fuse_names[idx], value))
    if lock_fuse:
[... 319 lines cut ...]
.debug_str                  3973         0
Total                      24037
<lambda>(["upload"], [".pio/build/ATmega3209_optiboot/firmware.hex"])
AVAILABLE: arduino
CURRENT: upload_protocol = arduino
BeforeUpload(["upload"], [".pio/build/ATmega3209_optiboot/firmware.hex"])
avrdude -v -p atmega3209 -C "/Users/hans/.platformio/packages/tool-avrdude-megaavr/avrdude.conf" -c arduino -b 115200 -P usb -U flash:w:.pio/build/ATmega3209_optiboot/firmware.hex:i

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/hans/.platformio/packages/tool-avrdude-megaavr/avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: ser_open(): can't open device "usb": No such file or directory

avrdude done.  Thank you.

*** [upload] Error 1
```

MCUdude, 2020-11-02:
@valeros any thoughts?

valeros, 2020-11-03:
Hi @MCUdude ! I've pushed your development with slight modification, including fixes for the issues your specified above. It'd be great if you could test new functionality using the platform for dev branch. Thanks!

MCUdude, 2020-11-03:
I'm not completely done testing, since there are some issues that are preventing me. First, I'm getting the incorrect device signature in Avrdude again. I've manually deleted the `tool-avrdude-megaavr` package. Here's the output:

```
$ pio run -t fuses -e fuses_bootloader
Processing fuses_bootloader (platform: https://github.com/platformio/platform-atmelmegaavr.git#develop; board: ATmega3209)
--------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/tool-avrdude-megaavr @ ~1.60300.0
Tool Manager: tool-avrdude-megaavr @ 1.60300.191015 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/ATmega3209.html
PLATFORM: Atmel megaAVR (1.2.0+sha.daf7155) > ATmega3209
HARDWARE: ATMEGA3209 16MHz, 4KB RAM, 32KB Flash
PACKAGES: 
 - tool-avrdude-megaavr 1.60300.191015 (6.3.0) 
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode

TARGET CONFIGURATION:
------------------------
Target = atmega3209
Clock speed = 16000000L
Oscillator = internal
BOD level = 2.7v
Save EEPROM = yes
Reset pin mode = reset
------------------------

Selected fuses:
------------------------
[fuse0 / WDTCFG = 0x0]
[fuse1 / BODCFG = 0x0]
[fuse2 / OSCCFG = 0x1]
[fuse4 / TCD0CFG = 0x0]
[fuse5 / SYSCFG0 = 0xc9]
[fuse6 / SYSCFG1 = 0x6]
[fuse7 / APPEND = 0x0]
[fuse8 / BOOTEND = 0x2]
[lfuse / LOCKBIT = 0xc5]
------------------------

Setting fuses...

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.07s

avrdude: Device signature = 0x1e9531
avrdude: Expected signature for ATmega3209 is 1E 95 53
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.

*** [fuses] Error 1
```

I'm not sure what's causing this. I was able to sneak in the -F flag to continue my testing.

Setting fuses works fine. However, bootloading not so much:

```
error: Couldn't find bootloader image bootloaders/bootloaders/optiboot/bootloaders/mega0/115200/Optiboot_mega0_UART3_DEF_115200_A7.hex

```

The corerct path is `bootloaders/optiboot/bootloaders/mega0/115200/Optiboot_mega0_UART3_DEF_115200_A7.hex ` I believe.

Another thing. I'm pretty sure all fuses are supposed to be in capital letters. At least, that's what Microchip uses in Atmel Studio and in their datasheets

MCUdude, 2020-11-03:
About the 0x200 offset. I think we should use the `upload_protocol` value as the indicator, not `board_hardware.uart`. There's no real reason to specify what exact UART port on your target that's used for serial uploads.

```
[env:Upload_UART]
platform = https://github.com/platformio/platform-atmelmegaavr.git
framework = arduino
board = ATmega3209
board_build.f_cpu = 16000000L
board_hardware.bod = 4.3v
board_build.variant = 48pin-standard
upload_protocol = arduino
upload_port = /dev/cu.usbserial-1420
```

In this example, since `upload_protocol = arduino`, it means that we're using a bootloader, and thus 0x200 is needed. It's also a benefit since `upload_protocol ` has to be defined anyways if you're using a bootloader for uploads. Another 0x200 related bug; 0x200 is actually added when no bootloader is used, so it's the other way around. But again `upload_protocol` would be better

valeros, 2020-11-03:
> First, I'm getting the incorrect device signature in Avrdude again. I've manually deleted the tool-avrdude-megaavr package.

Thanks for testing! I see you didn't specified `framework = arduino` in your configuration. That explains most of the issues. I suppose we should allow using your boards without any framework? In this case. where users should get bootloader binaries? Arduino packages are optional, if users don't run any example with Arduino framework, they won't be downloaded.

> Another thing. I'm pretty sure all fuses are supposed to be in capital letters. At least, that's what Microchip uses in Atmel Studio and in their datasheets.

Thanks, now I see why you used `%.2X`, should be fixed now.

> About the 0x200 offset. I think we should use the upload_protocol value as the indicator, not board_hardware.uart.

Thanks, fixed as well.

MCUdude, 2020-11-03:
> I see you didn't specified framework = arduino in your configuration. That explains most of the issues. 

🤦 Of course! I'm being sloppy, sorry! This fixed the issues I faced.

There's one tiny formatting thing I'd like to have fixed:

```
Selected fuses:
------------------------
[fuse0 / WDTCFG = 0x00]
[fuse1 / BODCFG = 0x00]
[fuse2 / OSCCFG = 0x01]
[fuse4 / TCD0CFG = 0x00]
[fuse5 / SYSCFG0 = 0xC9]
[fuse6 / SYSCFG1 = 0x06]
[fuse7 / APPEND = 0x00]
[fuse8 / BOOTEND = 0x02]
[lfuse / LOCKBIT = 0xc5]
------------------------
```

You can probably guess what it is? Add a space in front of the `=` character to get everything aligned 👍

MCUdude, 2020-11-04:
Another question @valeros. Is it possible to get access to a parameter defined in one environment in another environment?
Take this example:

```ini

[env:environment1]
upload_protocol = xplainedmini_updi

[env:environment2]
upload_protocol = ${environment1.upload_protocol}

```
This would be very convenient in the platformio.ini template I'm creating for MegaCoreX

valeros, 2020-11-04:
Thanks, the formatting should be fine now.

MCUdude, 2020-11-04:
Awesome!

Here are the README and platformio.ini template I've created for MegaCoreX users:

https://github.com/MCUdude/MegaCoreX/blob/master/PlatformIO.md

MCUdude, 2020-11-04:
Are we ready for a new atmelmegaavr release?

closed 2020-11-04 by MCUdude

valeros, 2020-11-04:
> Here are the README and platformio.ini template I've created for MegaCoreX users:

Is there any chance you could port that page to PlatformIO docs?  Something similar to the `atmelavr` docs you added earlier.

> Are we ready for a new atmelmegaavr release?

Once the issue #10 is resolved.

MCUdude, 2020-11-04:
> Once the issue #10 is resolved.

Sure! I do have an Uno Wifi Rev2, so iI can help out with the testing this evening.

MCUdude, 2020-11-09:
@valeros are we ready for a new release? I don't think there will be any significant changes in this repo for a while since pretty much all MegaCoreX features are implemented, and the bugs regarding Uno Wifi Rev2/Nano Every (#10).

valeros, 2020-11-09:
Thanks, released.