platformio/platform-atmelmegaavr #10: Unable to flash Arduino Uno WiFi Rev2

bsodmike, 2020-11-03:
What kind of issue is this?

- [X] **PlatformIO ATmelAVR**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**:
Mac OS 10.14.6

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 5.0.2

### Description of problem

Flashing the MCU with Platformio causes the built-in LED to simply keep flashing, and the flashed code has no effect.

#### Steps to Reproduce

Here's the code used to flash to the board, named as `src/main.cpp`:

```cpp
#include "Arduino.h"

#define BUILT_IN_LED LED_BUILTIN
#define PULSE_DELAY 500

void setupPulseLED(void)
{
  pinMode(BUILT_IN_LED, OUTPUT);
  digitalWrite(BUILT_IN_LED, LOW);
}

void pulse(void)
{
  digitalWrite(BUILT_IN_LED, HIGH);
  delay(PULSE_DELAY);
  digitalWrite(BUILT_IN_LED, LOW);
  delay(PULSE_DELAY);
}

void setup()
{
  setupPulseLED();
  Serial.begin(115200);
}

void loop()
{
  pulse();
  Serial.print("loop...");
}
```

Here's the upload:

```
Processing uno_wifi_rev2 (platform: atmelmegaavr; board: uno_wifi_rev2; framework: arduino)
---------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/uno_wifi_rev2.html
PLATFORM: Atmel megaAVR (1.2.0) > Arduino Uno WiFi Rev2
HARDWARE: ATMEGA4809 16MHz, 6KB RAM, 47.50KB Flash
PACKAGES:
 - framework-arduino-megaavr 1.8.6
 - tool-avrdude-megaavr 1.60300.191015 (6.3.0)
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
[... 190 lines cut ...]
}

void pulse(void)
{
  digitalWrite(BUILT_IN_LED, HIGH);
  delay(PULSE_DELAY);
  digitalWrite(BUILT_IN_LED, LOW);
  delay(PULSE_DELAY);
}

void setup()
{
  setupPulseLED();
  Serial.begin(115200);
}

void loop()
{
  pulse();
  Serial.print("loop...");
}

```

### Additional info

valeros, 2020-11-03:
Hi @bsodmike ! Thanks for reporting. Could you please try to upload using the upstream version of the platform? Just use the following configuration:
```ini
[env:uno_wifi_rev2]
platform = https://github.com/platformio/platform-atmelmegaavr.git
framework = arduino
board = uno_wifi_rev2
```

bsodmike, 2020-11-04:
Hi, thanks for your help!

With a working sketch on the board, I tried to upload another one to just blink the main LED with 500ms delay.  Instead, this seems to revert the LED to blink rapidly and simply not respond.  I wonder if it's wiping the bootloader or placing it some odd state.

In any case, behavior is still the same.

```
pio run -t upload
Processing uno_wifi_rev2 (platform: https://github.com/platformio/platform-atmelmegaavr.git; framework: arduino; board: uno_wifi_rev2)
---------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/uno_wifi_rev2.html
PLATFORM: Atmel megaAVR (1.2.0+sha.3c2f7d8) > Arduino Uno WiFi Rev2
HARDWARE: ATMEGA4809 16MHz, 6KB RAM, 47.50KB Flash
PACKAGES:
 - framework-arduino-megaavr 1.8.6
 - tool-avrdude-megaavr 1.60300.191015 (6.3.0)
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Selected fuses:
[fuse2 = 0x01]
[fuse5 = 0xc9]
[fuse8 = 0x02]
Compiling .pio/build/uno_wifi_rev2/src/main.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduinoVariant/variant.c.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/CDC.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/NANO_compat.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/UART.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/UART0.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/UART1.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/UART2.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/UART3.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/UNO_compat.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/USBCore.cpp.o
Archiving .pio/build/uno_wifi_rev2/libFrameworkArduinoVariant.a
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/WInterrupts.cpp.o
Indexing .pio/build/uno_wifi_rev2/libFrameworkArduinoVariant.a
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/abi.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/api/Common.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/api/IPAddress.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/api/PluggableUSB.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/api/Print.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/api/Stream.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/api/String.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/hooks.c.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/main.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/new.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/wiring.c.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/wiring_analog.c.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/wiring_digital.c.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/wiring_pulse.S.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/uno_wifi_rev2/libFrameworkArduino.a
Indexing .pio/build/uno_wifi_rev2/libFrameworkArduino.a
Linking .pio/build/uno_wifi_rev2/firmware.elf
Building .pio/build/uno_wifi_rev2/firmware.hex
Checking size .pio/build/uno_wifi_rev2/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   2.8% (used 175 bytes from 6144 bytes)
Flash: [=         ]   5.8% (used 2843 bytes from 48640 bytes)
Configuring upload protocol...
AVAILABLE: xplainedmini_updi
[... 88 lines cut ...]
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/Users/mdesilva/.platformio/packages/framework-arduino-megaavr/bootloaders/atmega4809_uart_bl.hex"
avrdude: writing flash (280 bytes):

Writing | ################################################## | 100% 0.12s

avrdude: 280 bytes of flash written
avrdude: verifying flash memory against /Users/mdesilva/.platformio/packages/framework-arduino-megaavr/bootloaders/atmega4809_uart_bl.hex:
avrdude: load data flash data from input file /Users/mdesilva/.platformio/packages/framework-arduino-megaavr/bootloaders/atmega4809_uart_bl.hex:
avrdude: input file /Users/mdesilva/.platformio/packages/framework-arduino-megaavr/bootloaders/atmega4809_uart_bl.hex contains 280 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.10s

avrdude: verifying ...
avrdude: 280 bytes of flash verified

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)

avrdude done.  Thank you.

====================================== [SUCCESS] Took 3.77 seconds ======================================
```

bsodmike, 2020-11-04:
@valeros I noticed something odd in the logs.  It's uploading my firmware, then setting the fuses, THEN erasing the chip and flashing the bootloader.  Shouldn't the bootloader get flashed first onto the chip?

bsodmike, 2020-11-04:
Here's some progress. I first set about uninstalling all installations of "atmelmegaavr" - locally I had two.  The stable release and the latest upstream in github.

My previous comment is valid in the sense, if you check my log below, the compiled firmware is now uploaded AFTER the bootloader is flashed to the chip.

However, behavior is different this time - the built-in LED is just "ON" and does not pulse.
No serial communications either.

Thoughts?

```
➜  iot-arduino-wifi-rev2 git:(master) ✗ pio platform uninstall "atmelmegaavr"
Platform Manager: Removing atmelmegaavr @ 1.2.0
Platform Manager: atmelmegaavr @ 1.2.0 has been removed!
Tool Manager: Removing framework-arduino-megaavr @ 1.8.6
Tool Manager: framework-arduino-megaavr @ 1.8.6 has been removed!
Tool Manager: Removing tool-avrdude-megaavr @ 1.60300.191015
Tool Manager: tool-avrdude-megaavr @ 1.60300.191015 has been removed!
Tool Manager: Removing toolchain-atmelavr @ 1.70300.191015
Tool Manager: toolchain-atmelavr @ 1.70300.191015 has been removed!
The platform 'atmelmegaavr' has been successfully removed!
➜  iot-arduino-wifi-rev2 git:(master) ✗ pio platform uninstall "atmelmegaavr"
Error: Could not find the package with 'atmelmegaavr' requirements for your system 'darwin_x86_64'
➜  iot-arduino-wifi-rev2 git:(master) ✗ pio run -t upload
Processing uno_wifi_rev2 (platform: https://github.com/platformio/platform-atmelmegaavr.git; framework: arduino; board: uno_wifi_rev2)
---------------------------------------------------------------------------------------------------------
Platform Manager: Installing git+https://github.com/platformio/platform-atmelmegaavr.git
git version 2.20.1 (Apple Git-117)
Cloning into '/Users/mdesilva/.platformio/.cache/tmp/pkg-installing-0mhwhh7l'...
remote: Enumerating objects: 61, done.
remote: Counting objects: 100% (61/61), done.
remote: Compressing objects: 100% (45/45), done.
remote: Total 61 (delta 20), reused 27 (delta 5), pack-reused 0
Unpacking objects: 100% (61/61), done.
Platform Manager: atmelmegaavr @ 1.2.0+sha.830fb85 has been installed!
The platform 'https://github.com/platformio/platform-atmelmegaavr.git' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.
Tool Manager: Installing platformio/toolchain-atmelavr @ ~1.70300.0
Unpacking  [####################################]  100%
Tool Manager: toolchain-atmelavr @ 1.70300.191015 has been installed!
Tool Manager: Installing platformio/framework-arduino-megaavr @ ~1.8.6
Tool Manager: framework-arduino-megaavr @ 1.8.6 has been installed!
Tool Manager: Installing platformio/tool-avrdude-megaavr @ ~1.60300.0
Tool Manager: tool-avrdude-megaavr @ 1.60300.191015 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/uno_wifi_rev2.html
PLATFORM: Atmel megaAVR (1.2.0+sha.830fb85) > Arduino Uno WiFi Rev2
HARDWARE: ATMEGA4809 16MHz, 6KB RAM, 47.50KB Flash
PACKAGES:
 - framework-arduino-megaavr 1.8.6
 - tool-avrdude-megaavr 1.60300.191015 (6.3.0)
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 7 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/uno_wifi_rev2/src/main.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduinoVariant/variant.c.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/CDC.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/NANO_compat.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/Tone.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/UART.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/UART0.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/UART1.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/UART2.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/UART3.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/UNO_compat.cpp.o
Compiling .pio/build/uno_wifi_rev2/FrameworkArduino/USBCore.cpp.o
Archiving .pio/build/uno_wifi_rev2/libFrameworkArduinoVariant.a
[... 103 lines cut ...]
avrdude: writing flash (280 bytes):

Writing | ################################################## | 100% 0.12s

avrdude: 280 bytes of flash written
avrdude: verifying flash memory against /Users/mdesilva/.platformio/packages/framework-arduino-megaavr/bootloaders/atmega4809_uart_bl.hex:
avrdude: load data flash data from input file /Users/mdesilva/.platformio/packages/framework-arduino-megaavr/bootloaders/atmega4809_uart_bl.hex:
avrdude: input file /Users/mdesilva/.platformio/packages/framework-arduino-megaavr/bootloaders/atmega4809_uart_bl.hex contains 280 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.10s

avrdude: verifying ...
avrdude: 280 bytes of flash verified
avrdude: reading input file ".pio/build/uno_wifi_rev2/firmware.hex"
avrdude: writing flash (3356 bytes):

Writing | ################################################## | 100% 0.78s

avrdude: 3356 bytes of flash written

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)

avrdude done.  Thank you.
```

valeros, 2020-11-04:
@bsodmike Could  you please also paste here the upload log in verbose mode `pio run -t upload -v`?

MCUdude, 2020-11-04:
@valeros I do have an Arduino Uno Wifi Rev2 I can test with.

Here is the output of `pio run -t upload -v`. The onboard LED is supposed to toggle every second, but instead it's flashing very fast. So far I have to shake the board in order for my eyes to see it.

```ini
[env:uno_wifi_rev2]
platform = https://github.com/platformio/platform-atmelmegaavr.git#develop
framework = arduino
board = uno_wifi_rev2
```

```
$ pio run -t upload -v
Processing uno_wifi_rev2 (platform: https://github.com/platformio/platform-atmelmegaavr.git#develop; framework: arduino; board: uno_wifi_rev2)
--------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/uno_wifi_rev2.html
PLATFORM: Atmel megaAVR (1.2.0+sha.e7000ef) (git+https://github.com/platformio/platform-atmelmegaavr.git#develop) > Arduino Uno WiFi Rev2
HARDWARE: ATMEGA4809 16MHz, 6KB RAM, 47.50KB Flash
PACKAGES: 
 - framework-arduino-megaavr 1.8.6 
 - tool-avrdude-megaavr 1.60300.191015 (6.3.0) 
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
MethodWrapper(["checkprogsize"], [".pio/build/uno_wifi_rev2/firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.4% (used 22 bytes from 6144 bytes)
Flash: [          ]   2.3% (used 1122 bytes from 48640 bytes)
.pio/build/uno_wifi_rev2/firmware.elf  :
section                     size      addr
.data                          4   8398848
.text                       1118       512
.bss                          18   8398852
.comment                      17         0
.note.gnu.avr.deviceinfo      64         0
.debug_aranges               224         0
.debug_info                10626         0
.debug_abbrev               9012         0
.debug_line                 1593         0
.debug_str                  3979         0
Total                      26655
<lambda>(["upload"], [".pio/build/uno_wifi_rev2/firmware.hex"])
AVAILABLE: xplainedmini_updi
CURRENT: upload_protocol = xplainedmini_updi
BeforeUpload(["upload"], [".pio/build/uno_wifi_rev2/firmware.hex"])
avrdude -v -p atmega4809 -C "/Users/hans/.platformio/packages/tool-avrdude-megaavr/avrdude.conf" -c xplainedmini_updi -Ufuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x02:m -Ulock:w:0xc5:m -Uflash:w:"/Users/hans/.platformio/packages/framework-arduino-megaavr/bootloaders/atmega4809_uart_bl.hex":i -b 115200 -V -e -D -P usb -U flash:w:.pio/build/uno_wifi_rev2/firmware.hex:i

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/hans/.platformio/packages/tool-avrdude-megaavr/avrdude.conf"
         User configuration file is "/Users/hans/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : xplainedmini_updi
         Overriding Baud Rate          : 115200
avrdude: Found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : ATmega4809
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
[... 113 lines cut ...]

Writing | ################################################## | 100% 0.12s

avrdude: 280 bytes of flash written
avrdude: verifying flash memory against /Users/hans/.platformio/packages/framework-arduino-megaavr/bootloaders/atmega4809_uart_bl.hex:
avrdude: load data flash data from input file /Users/hans/.platformio/packages/framework-arduino-megaavr/bootloaders/atmega4809_uart_bl.hex:
avrdude: input file /Users/hans/.platformio/packages/framework-arduino-megaavr/bootloaders/atmega4809_uart_bl.hex contains 280 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.10s

avrdude: verifying ...
avrdude: 280 bytes of flash verified
avrdude: reading input file ".pio/build/uno_wifi_rev2/firmware.hex"
avrdude: writing flash (1634 bytes):

Writing | ################################################## | 100% 0.21s

avrdude: 1634 bytes of flash written

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)

avrdude done.  Thank you.

```

MCUdude, 2020-11-04:
Just to narrow it down I compiled a blink sketch in Arduino IDE and uploaded it to the Uno Wifi Rev2. Everything worked as expected. Here is the upload command:

```
/Users/hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/Users/hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega4809 -cxplainedmini_updi -Pusb -b115200 -e -D -Uflash:w:/var/folders/6l/ypg6qbw172v1s4vtt6g990tw0000gn/T/arduino_build_594744/Blink.ino.hex:i -Ufuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x02:m -Uflash:w:/Users/hans/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/bootloaders/atmega4809_uart_bl.hex:i 
```

I compiled the same sketch in PlatformIO, and used the exact same upload command, but only replaced the hex file. However, this doesn't work!
```
/Users/hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/Users/hans/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/etc/avrdude.conf -v -patmega4809 -cxplainedmini_updi -Pusb -b115200 -e -D -Uflash:w:.pio/build/uno_wifi_rev2/firmware.hex:i -Ufuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x02:m -Uflash:w:/Users/hans/Library/Arduino15/packages/arduino/hardware/megaavr/1.8.6/bootloaders/atmega4809_uart_bl.hex:i 
```

I also tried to build with MegaCoreX and the `uno-wifi` pinout. This compiled blink sketch runs just fine as long as I manually specify `board_build.text_section_start = 0x200` in the platformio.ini file. To me, it seems like there is something wrong with the hex file being generated when compiling for the "original" Uno Wifi Rev2 target.

valeros, 2020-11-04:
Thanks for testing, it seems weak symbols from variant folder are getting overridden. Could you please pull the latest changes and try again?

MCUdude, 2020-11-04:
> Thanks for testing, it seems weak symbols from variant folder are getting overridden. Could you please pull the latest changes and try again?

That did the trick for the Nano Every and Uno Wifi Rev2. But will this affect any of the MegaCoreX targets?

valeros, 2020-11-04:
> But will this affect any of the MegaCoreX targets in any way?

You don't have source files in the `variant` folder, so it shouldn't have any impact on MegasCoreX.

MCUdude, 2020-11-04:
> You don't have source files in the variant folder, so it shouldn't have any impact on MegaCoreX.

Awesome! I've tested on actual hardware, and I can confirm this is the case too 🙂 

Thanks for doing this!

valeros, 2020-11-04:
Resolved in https://github.com/platformio/platform-atmelmegaavr/commit/cc8349d08bf9e5c6c7af9e8a47e399c5401da2db

closed 2020-11-04 by valeros

bsodmike, 2020-11-05:
Thanks I will test this tonight!