platformio/platform-atmelmegaavr #11: support for AVR Dx series / DxCore

sstaub, 2020-11-12:
I know this is the wrong place, but can you add support for the DxCore similar to the MegaCoreX?

davija, 2020-12-31:
@MCUdude @SpenceKonde
What all do we need to support this?
I just had some boards made and I'd much prefer to use Platform IO instead of Arduino IDE.

davija, 2021-01-01:
I'd like to help out with this, but I'm just not sure where to get started.  I was looking at getting the manifest files and stuff added, however it appears that a package needs to be created for the dxcore project and I'm not sure where to even get started.

MCUdude, 2021-01-01:
It was me and @valeros who added support for [MegaCoreX](https://github.com/MCUdude/MegaCoreX) and [megaTinyCore](https://github.com/SpenceKonde/megaTinyCore). Basically, what needs to be done is to create manifest files for each AVR-DA and AVR-DB part, and make sure all build parameters match the ones in the DxCore boards.txt file. Also, the fuses.py script needs to be updated.

I can do all this and submit a PR, but I'd like to know if @valeros have time for this now before I start working. He has been busy recently and not responded to #17 yet, which IMO is vital for DxCore support.

matou78, 2021-11-25:
hello :)
Sorry to bother you guys, but is there any news ? tried to install https://github.com/brunob45/platform-atmelavrdx to use the avr64DB32 but a lot of missing files :x

`Verbose` mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavrdx/AVR64DA32.html
PLATFORM: Atmel AVR Dx (1.5.0+sha.8ac103e) > AVR64DA32
HARDWARE: AVR64DA32 24MHz, 8KB RAM, 64KB Flash
PACKAGES:
 - framework-arduino-megaavr-dxcore 1.3.9+sha.7e1ed73
 - toolchain-atmelavr 2.70300.201015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 17 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <RF24> 1.4.1
|   |-- <SPI> 1.0.2
|-- <Arduino-Heartbeat-LED>
|-- <EEPROM> 2.1.1
|-- <SPI> 1.0.2
Building in release mode
Compiling .pio\build\avrda\src\main.cpp.o
Compiling .pio\build\avrda\lib386\SPI\SPI.cpp.o
avr-g++: error: device-specs/specs-avr64da32: No such file or directory
Compiling .pio\build\avrda\libae3\RF24\RF24.cpp.o
avr-g++: error: device-specs/specs-avr64da32: No such file or directory
Compiling .pio\build\avrda\libee3\Arduino-Heartbeat-LED\HeartbeatLed.cpp.o
Compiling .pio\build\avrda\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\avrda\FrameworkArduino\UART.cpp.o
Compiling .pio\build\avrda\FrameworkArduino\UART0.cpp.o
avr-g++: error: device-specs/specs-avr64da32: No such file or directory
Compiling .pio\build\avrda\FrameworkArduino\UART1.cpp.o
avr-g++: error: device-specs/specs-avr64da32: No such file or directory
Compiling .pio\build\avrda\FrameworkArduino\UART2.cpp.o
Compiling .pio\build\avrda\FrameworkArduino\UART3.cpp.o
avr-g++: error: device-specs/specs-avr64da32: No such file or directory
*** [.pio\build\avrda\src\main.cpp.o] Error 1
Compiling .pio\build\avrda\FrameworkArduino\UART4.cpp.o
*** [.pio\build\avrda\lib386\SPI\SPI.cpp.o] Error 1
*** [.pio\build\avrda\libae3\RF24\RF24.cpp.o] Error 1
*** [.pio\build\avrda\libee3\Arduino-Heartbeat-LED\HeartbeatLed.cpp.o] Error 1
Compiling .pio\build\avrda\FrameworkArduino\UART5.cpp.o
avr-g++: error: device-specs/specs-avr64da32: No such file or directory
*** [.pio\build\avrda\FrameworkArduino\Tone.cpp.o] Error 1
*** [.pio\build\avrda\FrameworkArduino\UART.cpp.o] Error 1
avr-g++: error: device-specs/specs-avr64da32: No such file or directory
*** [.pio\build\avrda\FrameworkArduino\UART0.cpp.o] Error 1
avr-g++: error: device-specs/specs-avr64da32: No such file or directory
avr-g++: error: device-specs/specs-avr64da32: No such file or directory
*** [.pio\build\avrda\FrameworkArduino\UART1.cpp.o] Error 1
*** [.pio\build\avrda\FrameworkArduino\UART2.cpp.o] Error 1
avr-g++: error: device-specs/specs-avr64da32: No such file or directory
avr-g++: error: device-specs/specs-avr64da32: No such file or directory
*** [.pio\build\avrda\FrameworkArduino\UART4.cpp.o] Error 1
*** [.pio\build\avrda\FrameworkArduino\UART3.cpp.o] Error 1
avr-g++: error: device-specs/specs-avr64da32: No such file or directory
*** [.pio\build\avrda\FrameworkArduino\UART5.cpp.o] Error `1`

SpenceKonde, 2021-11-25:
That looks like the last Official Arduino toolchain version, The one I've been distributing with DxCore and megaTinyCore works up through DD64's which aren't even out. 
```
{
          "name": "avr-gcc",
          "version": "7.3.0-atmel3.6.1-azduino4b",
          "systems": [
            {
              "size": "41351111",
              "checksum": "SHA-256:a0b8a6327a4ce0b2aee3e559a71ea0b61ccecc48e358ec5fac50ba10710981a0",
              "host": "i686-mingw32",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino4b-i686-w64-mingw32.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino4b-i686-w64-mingw32.tar.bz2"
            },
            {
              "size": "37878659",
              "checksum": "SHA-256:23becaaf746e8cf3b23ebcd261bf52d5eebcc094c3500fe8d533f0911b870c60",
              "host": "aarch64-linux-gnu",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino4a-aarch64-pc-linux-gnu.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino4a-aarch64-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "34220996",
              "checksum": "SHA-256:dacef5c90b38e6728e2cd161a54133b008352ff6f6513e67ab71f3117331c758",
              "host": "arm-linux-gnueabihf",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino4a-arm-linux-gnueabihf.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino4a-arm-linux-gnueabihf.tar.bz2"
            },
            {
              "size": "36887243",
              "checksum": "SHA-256:a3f4833034b5526a7375b20fed7031e5fd0ff6956703e7b4bcf48e914fb94d6c",
              "host": "i686-linux-gnu",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino4a-i686-pc-linux-gnu.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino4a-i686-pc-linux-gnu.tar.bz2"
            },
            {
              "size": "37530243",
              "checksum": "SHA-256:61c9ad88f9e75664fc6b96350dd4b3f95946e1c6eeca2158504aecc515f475a3",
              "host": "x86_64-apple-darwin14",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino4a-x86_64-apple-darwin14.tar.bz2",
              "url": "https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino4a-x86_64-apple-darwin14.tar.bz2"
            },
            {
              "size": "37412841",
              "checksum": "SHA-256:60d879e7052db01383cd89c54e46fa20c81ca180e20a8c0a681ffecec20c24d8",
              "host": "x86_64-linux-gnu",
              "archiveFileName": "avr-gcc-7.3.0-atmel3.6.1-azduino4a-x86_64-pc-linux-gnu.tar.bz2",
              "url": "https://github.com/SpenceKonde/DxCore/releases/download/1.3.4/avr-gcc-7.3.0-atmel3.6.1-azduino4a-x86_64-pc-linux-gnu.tar.bz2"
            }
          ]
        }
```

matou78, 2021-11-25:
looks like I'm doing something wrong then in platformio
on launch of my workspace i get 

> [25/11/2021, 16:55:38] Unable to resolve configuration with compilerPath "C:/Users/Matou/.platformio/packages/toolchain-atmelavr/bin/avr-gcc.exe".  Using "cl.exe" instead.

i think will finally go to arduino ide to play with the new avr Dx chips :(

thanks for the response SpenceKonde

MCUdude, 2022-01-17:
@valeros this PR can be closed now that #19 has been merged.

valeros, 2022-01-18:
Resolved in https://github.com/platformio/platform-atmelmegaavr/pull/19

closed 2022-01-18 by valeros