platformio/platform-atmelmegaavr #15: Can't upload to nano every

gurghet, 2020-11-25:
I'm having a similar issue to https://github.com/platformio/platform-atmelmegaavr/issues/6.
Uploading from Arduino IDE works fine but PlatformIO hangs forever.
Following is my verbose shell.
```
PS C:\Users\gurgh\OneDrive\Documents\PlatformIO\Projects\Audiuino> pio run -t upload -v
Processing nano_every (platform: atmelmegaavr; board: nano_every; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------

CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/nano_every.html
PLATFORM: Atmel megaAVR (1.3.0) > Arduino Nano Every
HARDWARE: ATMEGA4809 16MHz, 6KB RAM, 47.50KB Flash
PACKAGES:
 - framework-arduino-megaavr 1.8.6
 - tool-avrdude-megaavr 1.60300.191015 (6.3.0) 
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Auto-detected: COM8
MethodWrapper(["checkprogsize"], [".pio\build\nano_every\firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.4% (used 26 bytes from 6144 bytes)
Flash: [          ]   3.0% (used 1456 bytes from 48640 bytes)
.pio\build\nano_every\firmware.elf  :

section                     size      addr

.data                          6   8398848

.text                       1346         0

.rodata                      104     17730

.bss                          20   8398854

.comment                      17         0

.note.gnu.avr.deviceinfo      64         0

.debug_aranges               192         0

.debug_info                10525         0

.debug_abbrev               8992         0

.debug_line                 1303         0

.debug_str                  3979         0

Total                      26548
<lambda>(["upload"], [".pio\build\nano_every\firmware.hex"])
AVAILABLE: jtag2updi
CURRENT: upload_protocol = jtag2updi
BeforeUpload(["upload"], [".pio\build\nano_every\firmware.hex"])
Use manually specified: COM8
Forcing reset using 1200bps open/close on port COM8
Waiting for the new upload port...
```
After a few minutes it will print
```
avrdude -v -p atmega4809 -C "C:\Users\gurgh\.platformio\packages\tool-avrdude-megaavr\avrdude.conf" -c jtag2updi -Ufuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x00:m -Ulock:w:0xc5:m -b 115200 -V -e -D -P "COM8" -U flash:w:.pio\build\nano_every\firmware.hex:i

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\gurgh\.platformio\packages\tool-avrdude-megaavr\avrdude.conf"

         Using Port                    : COM8
         Using Programmer              : jtag2updi
         Overriding Baud Rate          : 115200
```

chrisinabox, 2020-11-25:
I'm no expert but based on the fix that @valeros applied in issue #6 it seems that the changes made (commit 9ceb183) were accidentally removed in commit 89d5e79 when the the uploader configuration was changed to more generically handle jtag2updi online 200 of builder/main.py

`# jtag2updi seems to be the only protocol that requires serial port
    if upload_protocol == "jtag2updi":
        upload_options = env.BoardConfig().get("upload", {})
        for opt in ("require_upload_port", "use_1200bps_touch", "wait_for_upload_port"):
            upload_options[opt] = True`

So in short - looks like the fix was undone which is why it isn't working in the dev branch or master branch

gurghet, 2020-11-25:
Thanks, tonight I'll try and install https://github.com/platformio/platform-atmelmegaavr/commit/9ceb183865f622dc75fc5a1e32294a5e7ae358f1 and report if it works

gurghet, 2020-11-25:
Works perfectly! Thanks @chrisinabox 

Workaround: downgrade to 9ceb183

chrisinabox, 2020-11-25:
No worries, glad to help! 

Hopefully @valeros can take a look when they have some spare time. Clearly just a regression issue.

valeros, 2020-11-27:
Thanks for reporting.
@MCUdude Could you please clarify whether  there is any difference in the uploading process using the `jtag2updi` protocol for your targets from PR #5 and Nano Every? I see you used `wait_for_upload_port`, but it breaks uploading to the Nano board. Thanks!

MCUdude, 2020-11-27:
@valeros I'll give it a go this evening. When I tried this on my mac, `wait_for_upload_port` worked just fine. I'll try it on a Windows machine as well.

It may be that the serial port actually doesn't disappear when it gets the 1200bps touch but only switches functionality internally (unlike the Arduino Leonardo). A "generic/DIY" jtag2updi programmer does not use and will not be affected by a 1200pbs touch, so if `wait_for_upload_port` has to be removed for Nano Every, it means that its onboard jtag2updi programmer works more similar to a "generic/DIY" version, which is a good thing!

MCUdude, 2020-11-27:
I just tested this on my mac and on my Windows PC, and uploading works fine on both with the latest release

MCUdude, 2020-11-29:
@valeros I can also mention that it didn't make a difference if I added or removed `wait_for_upload_port` to the nano_every.json manifest file or `board_upload.wait_for_upload_port`to the platformio.ini file. I still got "Waiting for the new upload port..." regardless; on both MacOS and Windows.

valeros, 2020-11-30:
@MCUdude Thanks for looking into it. Probably you need to disable `wait_for_upload_port` directly in the [main.py#L202](https://github.com/platformio/platform-atmelmegaavr/blob/develop/builder/main.py#L202) to properly test it. Thanks!

MCUdude, 2020-11-30:
@valeros ok, so basically replace

```py
for opt in ("require_upload_port", "use_1200bps_touch", "wait_for_upload_port"):
            upload_options[opt] = True
```

with

```py
        upload_options["require_upload_port"] = True
        upload_options["use_1200bps_touch"] = True
        upload_options["wait_for_upload_port"] = False
```

then. I'll give it a try later today!

MCUdude, 2020-11-30:
I can confirm that

```py
upload_options["require_upload_port"] = True
upload_options["use_1200bps_touch"] = True
upload_options["wait_for_upload_port"] = False
```

Works on both Windows and Mac.

closed 2020-11-30 by valeros

valeros, 2020-11-30:
@gurghet Should be fixed in the dev branch, could you please retest with the following configuration:
```ini
[env:upstream_develop]
platform = https://github.com/platformio/platform-atmelmegaavr.git
framework = arduino
board = nano_every
```

gurghet, 2020-12-01:
Works on my mac! 👍

commit db586f05 (2020-11-30): Don't wait for upload port after reset with jtag2updi // Resolve #15