platformio/platform-atmelmegaavr #16: Board request for TinyCore

prosperv, 2020-11-30:
https://www.tindie.com/products/xkimi/tiny-core-32-dev-board-attiny3217/
https://www.tindie.com/products/xkimi/tiny-core-16-dev-board-attiny1616/

https://raw.githubusercontent.com/xukangmin/TinyCore/master/avr/package/package_tinycore_index.json

MCUdude, 2020-11-30:
There is no reason why these boards can't be used with megaTinyCore when support for this core is ready.

valeros, 2020-11-30:
Hi @prosperv ! Could you please briefly explain the benefits of using `TinyCore` when there is already `megaTinyCore ` that supports that MCUs?

MCUdude, 2020-11-30:
You can find the source for TinyCore [here](https://github.com/xukangmin/TinyCore). It's is pretty much identical to megaTinyCore (but has not been updated recently), and the only difference I've been able to find is the Arduino pin mappings in `pins_arduino.h`. but one can always use the PIN macros to refer to pins: `digitalWrite(PIN_PA7, HIGH);`.

IMO it's not a good idea to add cores to PlatformIO that's based on someone else's core, but are not under active development and aren't even that popular compared to megaTinyCore, which supports _all_ tiny0/1's.