platformio/platform-atmelmegaavr #17: Add support for new UPDI programmers + extended UPDI protocol

MCUdude, 2020-12-15:
Someone [mentioned at the Avrfreaks forum](https://www.avrfreaks.net/forum/avr128da48-curiosity-nano-and-avrdude) that an Avrdude [patch from one of the Atmel/Microchip developers has been published](https://savannah.nongnu.org/patch/?10000).

It adds support for the PICKit4 and Microchip SNAP. But more importantly, it adds extended UPDI addressing support for devices with more than 48kB flash. This means that the AVR DA/DB/DD/EA series now can be used with official Microchip UPDI programmers and not just jtag2updi.

@valeros can you merge this patch so the next release of platform-atmelmegaavr will have this new functionality? Thanks!

MCUdude, 2020-12-28:
@valeros any change for this patch to be added? The Arduino developers have already added the patch to their Avrdude repo.

Oh, by the way, what's really left before a new release of this repo can be created? We now have support for a bunch of new ATtinys, but there's no release for them.

MCUdude, 2021-01-24:
BTW is there any reason why the atmel-megaavr package has its own avrdude version? I've already asked this question in [platform-atmelavr#229](https://github.com/platformio/platform-atmelavr/issues/229), but it's also relevant here.

Here's why this is important:
> Wouldn't it be a good idea to combine the atmelavr and the atmelmegaavr Avrdude versions? Currently, the atmelmegaavr version is more up to date. The reason why I'm stressing this is because only the atmelmegaavr version supports the new MPLAB SNAP and PICkit4 programmers in ISP mode, and support for the latest official programmers is always a good thing.

valeros, 2021-01-27:
Hi @MCUdude, terribly sorry for the late reply.  Finally, I managed to reserve some time for this platform. I'll need a little time to look into the issues and PR you've submitted. 

> @valeros can you merge this patch so the next release of platform-atmelmegaavr will have this new functionality? Thanks!
@valeros any change for this patch to be added? The Arduino developers have already added the patch to their Avrdude repo.

We don't have any infrastructure for compiling avrdude from scratch for all popular OS including SBC, so if there are already prepared binaries somewhere I can take a look.

> BTW is there any reason why the atmel-megaavr package has its own avrdude version? I've already asked this question in platform-atmelavr#229, but it's also relevant here.

I believe it's a legacy thing, the `atmelavr` was the first platform with the `avrdude` package. If I remember correctly, we use the same binaries shipped with Arduino and looks like it's a fork of the original avrdude with some additional patches, so we just don't want to introduce another possible source of issues to look into when a user reports that uploading doesn't work.

> Oh, by the way, what's really left before a new release of this repo can be created? We now have support for a bunch of new ATtinys, but there's no release for them.

It'd be great if we could release the platform this month with DxCore or without.

MCUdude, 2021-01-27:
> Hi @MCUdude, terribly sorry for the late reply. Finally, I managed to reserve some time for this platform. I'll need a little time to look into the issues and PR you've submitted.

@valeros thanks for replying. I should be available tonight if that helps? I'm usually also available in the evening after work

>>@valeros can you merge this patch so the next release of platform-atmelmegaavr will have this new functionality? Thanks!
>>@valeros any change for this patch to be added? The Arduino developers have already added the patch to their Avrdude repo.

>We don't have any infrastructure for compiling avrdude from scratch for all popular OS including SBC, so if there are already prepared binaries somewhere I can take a look.

This is the Avrdude repo Arduino uses. It's absolutely up to date. They also have binaries (that's also up to date. https://github.com/facchinm/avrdude

I'm not sure what the download URL for their Avrdude binaries are, but I'm using the current release here: [MegaCoreX boards manager json](https://github.com/MCUdude/MegaCoreX/blob/d2fb183d36d6dfea20736da315d177f3fb735746/package_MCUdude_MegaCoreX_index.json#L425-L427).

Again, what's good with this version is that it supports the latest official Microchip programmers for pretty much all targets. That is _fantastic_ when working on professional applications, which I often do.

> I believe it's a legacy thing, the atmelavr was the first platform with the avrdude package. If I remember correctly, we use the same binaries shipped with Arduino, and looks like it's a fork of the original avrdude with some additional Arduino-specific patches, so we just don't want to introduce another possible source of issues to look into when a user reports that uploading doesn't work.

I've been using the latest avrdude version for a while now (for both AVR-DA, megaAVR-0, and "classic" AVR's) and it works excellent on all targets. I wouldn't be afraid of distributing this version as the "main" Avrdude version.

Is there anything you need from me in order to continue working on this? Like I've already mentioned, I'm usually always available to answer messages, and I'm able to work on this in the evenings.

valeros, 2021-01-27:
Thanks, the avrdude packages have been updated.

closed 2021-01-27 by valeros

MCUdude, 2021-01-27:
Cool! Is there a link to a github like webside where I can look at the commit? And is the atmel avr version also updated?

valeros, 2021-01-27:
> Is there a link to a github like webside where I can look at the commit?

There is no commit as only the patch version has changed, so just run `pio platform update atmelmegaavr`.

> And is the atmel avr version also updated?

At the moment only packages for `atmelmegaavr`. I hope `avrdude` for atmelavr will be updates as well a bit later.

MCUdude, 2021-01-27:
@valeros I just had a look in the avrdude.conf file that was bundled with the latest PlatformIO Avrdude release, and the new ISP programmers are missing. Would be awesome if you could add these to Avrdude.conf (from line 1257):

```
programmer
    id    = "pickit4_isp";
    desc  = "MPLAB(R) PICkit 4 in ISP mode";
    type  = "jtagice3_isp";
    connection_type = usb;
    usbpid = 0x2177;
 ;

 programmer
    id    = "snap_isp";
    desc  = "MPLAB(R) SNAP in ISP mode";
    type  = "jtagice3_isp";
    connection_type = usb;
    usbpid = 0x217F, 0x2180, 0x2181;
 ;
```

valeros, 2021-01-27:
Is there a reason that they aren't included by default in `v6.3.0-arduino18`?

MCUdude, 2021-01-27:
> Is there a reason that they aren't included by default in v6.3.0-arduino18?

There are no good reason really. v6.3.0-arduino18 Support the SNAP and PICkit4 in both UPDI and ISP mode, so why not make Avrdude utilize this? I have tested both programmers with ISP and UPDI, and can confirm that they work flawlessly with avrdude-v6.3.0-arduino18.

I think the real reason is that the stock Arduino IDE (that only supports "classic" AVR boards) ships with an old Avrude version that doesn't support these new programmers. However, if you install the official Arduino Atmel megaAVR package, the "old" Avrdude version is replaced/overridden by the new v6.3.0-arduino18 version, even for "classic" AVR boards.

So there's no good reason why MPLAB SNAP and PICkit4 in ISP mode isn't added to avrdude.conf by default. Since my cores (including MegaCoreX) uses its own avrdude.conf, I don't have to worry about this for the Arduino installation. However, PlatformIO is different and uses a "global" avrdude.conf file.

valeros, 2021-01-27:
The packages have been updated with the configs for `pickit4_isp` and `snap_isp`.

MCUdude, 2021-01-27:
Excellent! 🎉