platformio/platform-atmelmegaavr #23: Feature Request: Please support DxCore by SpenceKonde

btribley, 2021-02-27:
- [ ] **Feature Request**.
 https://github.com/SpenceKonde/DxCore This is a "board" of bare chips perfect for people like me who make custom hardware. I made a board with the AVR128DA28 only to find out that the Microchip software support is all but broken for things like I2C. Arduino is a great hardware/software standard but their IDE is horribly basic. Their imports are all broken for any 3rd party core. PlatformIO would be a perfect IDE.

btribley, 2021-03-15:
SpenceKonde is up to version 1.3.2 . It's looking pretty good. Any progress on getting this into a build so it can be used?

MCUdude, 2022-01-17:
This issue can be closed now that #19 has been merged

valeros, 2022-01-18:
Resolved in https://github.com/platformio/platform-atmelmegaavr/pull/19