platformio/platform-atmelmegaavr #25: Support tinyAVR 2 series

gcormier, 2021-04-16:
With release 2.3.0, https://github.com/SpenceKonde/megaTinyCore supports the new TinyAVR series.

How can we get this in PIO?

mechatroniks-git, 2021-05-20:
ATtiny1624 chips (2-Series) have been out for a while, would be nice to make this happen.

5e3, 2021-06-17:
I build myself a dev-board with an attiny1626 (and another one one with a 128db28) and I was naively surprised that none of these processors are yet supported by Platformio. You know, its so painful to work with the Arduino toolset (frankly it's only good enough to toy around. When you need to know how something is defined you have to search the whole computer whereas with Platformio you easily get to the respective source just from the IDE). So please support the tinyAVR2 series (and the AVRxDB) series. 
I have no idea how the tool-chain is maintained but Spence Conde's core libraries come with a working tool-chain. I compared it with the current platformio megaavr tool-chain and it seems to suggest that it's still the 7.3 avr-gcc version extended with a small set of .h and .o files that where changed/created because of new mcu's.
I never engaged in a open source environment but please tell me how I can help to make this happen. (I saw there are PRs open since 6 months.)

marco-117, 2022-01-11:
Me too. 
I also need support of the Attiny 1624 in PlatformIO.

Is there any tutorial that i can implement this chip by myself?

MCUdude, 2022-01-17:
I'll create a PR soon with support for the entire tinyAVR-2 lineup. Stay tuned!

PR #36 (2022-01-18): Add support for the entire tinyAVR-2 family
Support for the entire tinyAVR-2 series is currently missing. This PR solved this.

Closes #25 
Closes https://github.com/platformio/platform-atmelavr/issues/268
Closes https://github.com/platformio/platform-atmelavr/issues/263
closed 2022-01-18 by valeros