platformio/platform-atmelmegaavr #26: Board macros have different value than expected

maxgerhardt, 2021-04-27:
See https://community.platformio.org/t/impossible-to-build-adafruit-gfx-with-platformio/21009/5?u=maxgerhardt. 

Situation: Attempt to compile a Adafruit GFX Sketch with the `platformio.ini`
```
[env:ATmega4809]
platform = atmelmegaavr
board = ATmega4809
framework = arduino
lib_ldf_mode = chain+
lib_deps =
     adafruit/Adafruit GFX Library @ ^1.10.7
     adafruit/Adafruit BusIO @ ^1.7.3
     SPI
     Wire
```

The Adafruit GFX Core tries to detect the ATMega4809 with [the code]

```
..  defined(ARDUINO_AVR_ATmega4809) || ...
```
However this core defines

https://github.com/platformio/platform-atmelmegaavr/blob/dcbaeb126497f08acf2914ce5b8ebb28de742d58/boards/ATmega4809.json#L4-L4

Notice the different upper/lowercase spelling.

This causes a compile failure for Adafruit GFX sketches which rely on this check to activate the definition of certain types (`BitOrder`). 

It should be double-checked whether PlatformIO makes a spelling mistake here and if the behavior in the MegaCoreX is the same.

maxgerhardt, 2021-04-27:
Maybe @MCUdude can check that with regards to the compatibility of the Adafruit GFX library with MegaCoreX? Original PR #2 with those macro names come from there.

MCUdude, 2021-04-28:
Hi @maxgerhardt!

You're absolutely right, it's supposed to be `ARDUINO_AVR_ATmega4809` rather than `AVR_ATMEGA4809`. I'll provide a PR that fixes this

valeros, 2021-04-28:
Resolved in https://github.com/platformio/platform-atmelmegaavr/pull/27

closed 2021-04-28 by valeros