platformio/platform-atmelmegaavr #31: avrdude: Can't find programmer id "curiosity_updi"

vishnumaiea, 2021-09-04:
**PlatformIO Core Version:** 5.1.1
**Home Version:** 3.3.4
**IDE:** VS Code
Version: 1.60.0 (user setup)
Commit: e7d7e9a9348e6a8cc8c03f877d39cb72e5dfb1ff
Date: 2021-09-01T10:41:52.311Z
Electron: 13.1.8
Chrome: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Windows_NT x64 10.0.19043

I was trying to build a sample project for **AVR-IoT WG** development board. PlatformIO supports the board, and builds the project using Arduino framework. My code (to blink an LED) compiles fine, but I am seeing issues while uploading. Following is the verbose output from Build command.

```
> Executing task: C:\Users\Vishnu Mohanan\.platformio\penv\Scripts\platformio.exe run --verbose --target upload --environment avr_iot_wg <

Processing avr_iot_wg (platform: atmelmegaavr; board: avr_iot_wg; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/avr_iot_wg.html
PLATFORM: Atmel megaAVR (1.5.0) > AVR-IoT WG Development Board
HARDWARE: ATMEGA4808 16MHz, 6KB RAM, 48KB Flash
PACKAGES:
 - framework-arduino-megaavr-megacorex 1.0.9
 - tool-avrdude-megaavr 2.60300.210128 (6.3.0)
 - toolchain-atmelavr 2.70300.201015 (7.3.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 11 compatible libraries
Scanning dependencies...     
No dependencies
Building in release mode
MethodWrapper(["checkprogsize"], [".pio\build\avr_iot_wg\firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.1% (used 4 bytes from 6144 bytes)
Flash: [          ]   1.6% (used 792 bytes from 49152 bytes)
.pio\build\avr_iot_wg\firmware.elf  :

section                     size      addr

.data                          0   8398848

.text                        792         0

.bss                           4   8398848

.comment                      17         0

.note.gnu.avr.deviceinfo      64         0

.debug_aranges               128         0

.debug_info                 9628         0

.debug_abbrev               8466         0

.debug_line                  921         0

.debug_str                  3789         0

Total                      23809
<lambda>(["upload"], [".pio\build\avr_iot_wg\firmware.hex"])
AVAILABLE: curiosity_updi
CURRENT: upload_protocol = curiosity_updi
BeforeUpload(["upload"], [".pio\build\avr_iot_wg\firmware.hex"])
avrdude -v -p atmega4808 -C "C:\Users\Vishnu Mohanan\.platformio\packages\tool-avrdude-megaavr\avrdude.conf" -c curiosity_updi -P usb -U flash:w:.pio\build\avr_iot_wg\firmware.hex:i

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

[... 143 lines cut ...]
Platform atmelavr
--------
Updating platformio/atmelavr                  3.4.0                              [Up-to-date]
Updating platformio/toolchain-atmelavr        1.70300.191015 @ ~1.70300.0        [Up-to-date]

Platform atmelmegaavr
--------
Updating platformio/atmelmegaavr              1.5.0                              [Up-to-date]
Updating platformio/toolchain-atmelavr        1.70300.191015 @ ~1.70300.0        [Up-to-date]
Updating platformio/framework-arduino-megaavr-megacorex 1.0.9 @ ~1.0.7                     [Up-to-date]

Library Manager
===============
Library Storage: C:\Users\Vishnu Mohanan\.platformio\lib
```

```
PS D:\Code\PlatformIO\AVR-IoT\Blink\AVR-IoT-Blink> pio upgrade
You're up-to-date!
PlatformIO 5.1.1 is currently the newest version available.
```

Is it not possible to upload binary to the board via USB port? My fallback option was to drag and drop the HEX file to the mass storage device appearing as **CURIOSITY**.

Please look into the issue and consider updating the support or suggesting a solution.

MCUdude, 2022-01-17:
@valeros can you add these missing programmers to avrdude.conf?

brunob45, 2022-01-18:
As of `tool-avrdude-megaavr 2.60300.210128 (6.3.0)`, pkobn_updi is present in both windows and linux versions.
In windows, I'm able to upload a sketch to my Curiosity Nano AVR128DA48 board with the package "as-is", but in linux, avrdude seems to be missing the dependency "libhidapi", and must be recompiled.
When I compile avrdude from [source](https://github.com/avrdudes/avrdude) on ubuntu with the package libhidapi-dev installed, the upload works fine.

PR #35 (2022-01-18): DxCore - add Curiosity Nano boards
I added the Curiosity Nano DA & DB board definitions.
- https://www.microchip.com/developmenttools/ProductDetails/DM164151
- https://www.microchip.com/developmenttools/ProductDetails/EV35L43A

I confirmed that the build and upload work on my Curiosity Nano AVR128DA48, both on windows (as-is) and linux (avrdude fixed #31).

Also, since MILLIS_USE_TIMER is defined in arduino.py, I removed the compiler flag from every .json board definitions, because they were conflicting.
closed 2022-01-18 by valeros