platformio/platform-atmelmegaavr #32: Natively support pymcuprogram

maxgerhardt, 2021-10-30:
The `microchip-pic-avr-tools/pymcuprog` has emerged and per https://github.com/microchip-pic-avr-tools/pymcuprog/blob/main/pypi.md#serial-port-updi-pyupdi is capable of uploading to a MegaAVR device via UPDI via a serial adapter, in the same way [pyupdi](https://github.com/mraardvark/pyupdi#pyupdi-1) can.

One might think integrating PyUPDI should be done, but as of 3 days a [deprecation notice](https://github.com/mraardvark/pyupdi#deprecation-notice) was posted pointing to the `pymcuprog` program as its successor. The [`serialupdi`](https://github.com/microchip-pic-avr-tools/pymcuprog/tree/main/pymcuprog/serialupdi) subfolder is [PyUPDI](https://github.com/mraardvark/pyupdi/tree/master/updi) slightly adapted.

As of now, PlatformIO only supported PyUPDI via `upload_protocol = custom` configurations anyways, not natively (see [here](https://github.com/MCUdude/MegaCoreX/blob/master/PlatformIO.md#pyupdi)).

Programming "raw" MegaAVR chips via a simple serial adapter is a common use-case for hobbyists bying the chips directly (such as the convienent DIP form the ATMega4809) and as such, PlatformIO should support uploading the firmware and setting fuses natively through using only a serial adapter and the new `pymcuprog` program.

MCUdude, 2022-01-05:
The next release of Avrdude will natively support SerialUPDI. It has already been pushed to the upstream repo.

https://github.com/avrdudes/avrdude

matou78, 2022-05-11:
just tested https://github.com/avrdudes/avrdude/releases/tag/v7.0 by replacing "avrdude.exe" and "avrdude.conf" inside of ".platformio\packages\tool-avrdude-megaavr" ( i'm using an AVR64DA32 ) and was able to use "upload_protocol = serialupdi"

mairas, 2023-01-31:
Avrdude 7.0 and 7.1 are indeed out and work reliably with serialupdi programmers. IMHO, this issue can be closed (and #46 should be fixed).

matou78, 2023-02-03:
for me all good and sexy :) 
[image]