platformio/platform-atmelmegaavr #33: Users can't upload to Uno WiFi Rev2 because of argument escaping problem

maxgerhardt, 2021-11-12:
With the `platformio.ini`

```ini
[env:uno_wifi_rev2
platform = atmelmegaavr
board = uno_wifi_rev2
framework = arduino
```

Users with a space in their name experience a failure in uploading.

```
<lambda>(["upload"], [".pio\build\uno_wifi_rev2\firmware.hex"])
AVAILABLE: xplainedmini_updi
CURRENT: upload_protocol = xplainedmini_updi
BeforeUpload(["upload"], [".pio\build\uno_wifi_rev2\firmware.hex"])
avrdude -v -p atmega4809 -C "C:\Users\Wong Zhi Wei\.platformio\packages\tool-avrdude-megaavr\avrdude.conf" -c xplainedmini_updi -Ufuse2:w:0x01:m -Ufuse5:w:0xC9:m -Ufuse8:w:0x02:m -Ulock:w:0xC5:m "-Uflash:w:"C:\Users\Wong Zhi Wei\.platformio\packages\framework-arduino-megaavr\bootloaders\atmega4809_uart_bl.hex":i" -b 115200 -V -e -D -P usb -U flash:w:.pio\build\uno_wifi_rev2\firmware.hex:i
avrdude: invalid file format '\Users\Wong' in update specifier
avrdude: error parsing update operation 'flash:w:C:\Users\Wong'
*** [upload] Error 1
```

The problem looks to me like there's a double-qupting of the flashing argument for the bootloader

>"-Uflash:w:"C:\Users\Wong Zhi Wei\.platformio\packages\framework-arduino-megaavr\bootloaders\atmega4809_uart_bl.hex":i"

Per [this](https://stackoverflow.com/questions/32658448/how-to-give-full-path-name-in-avrdude-command) the argument should have been just

>-Uflash:w:"C:\Users\Wong Zhi Wei\.platformio\packages\framework-arduino-megaavr\bootloaders\atmega4809_uart_bl.hex":i

I however don't see where those quotes for the complete command get added..

https://github.com/platformio/platform-atmelmegaavr/blob/365d8bac9ab853ac78a3e2ff54de100732e08ed5/builder/main.py#L214-L218

https://github.com/platformio/platform-atmelmegaavr/blob/365d8bac9ab853ac78a3e2ff54de100732e08ed5/builder/bootloader.py#L87-L87

only add quotes for the bootloader path, *which is correct*. There must be some auto-escaping screwing things up here.

valeros, 2021-11-16:
Hi @maxgerhardt ! Thanks for pointing out. I believe SCons wraps upload flags containing whitespaces, so we can rely on this functionality.

closed 2021-11-16 by valeros

maxgerhardt, 2021-11-16:
By using the `platformio.ini`

```ini
[env:uno_wifi_rev2]
platform = https://github.com/platformio/platform-atmelmegaavr.git
board = uno_wifi_rev2
framework = arduino

[platformio]
packages_dir = package with spaces
```

This has had the effect of changing the invocation from the form 

>"-Uflash:w:"C:\Users\Wong Zhi Wei\.platformio\packages\framework-arduino-megaavr\bootloaders\atmega4809_uart_bl.hex":i"

to 

>-U "flash:w:C:\Users\Max\temp\uno_wifi\package with spaces\framework-arduino-megaavr\bootloaders\atmega4809_uart_bl.hex:i"

I'm not sore of this is better or worse with the missing quotes in the path argument, but at least I don't get a 

>avrdude: error parsing update operation 'flash:w:C:\Users\Wong'

style error anymore, just that no device was found (since I don't have a Uno WiFi Rev2 connected). 

Maybe the original user can test this.

maxgerhardt, 2021-11-16:
[Confirmed fixed](https://community.platformio.org/t/platformio-upload-error-for-arduino-uno-wifi-rev2/24451/10?u=maxgerhardt).

commit 317e3e82 (2022-05-30): Better handling of  bootloader path in AVRDUDE upload flags

Resolve #33