platformio/platform-atmelmegaavr #40: running upstream versions

br74649, 2022-02-10:
Hi

I would love to get started trying out the new support for avr-2, i see there has been some activity in adding support thanks @MCUdude 

so as per the readme, i add 

[env:development]
platform = https://github.com/platformio/platform-atmelmegaavr.git
board = ...
...

to try and build a project for the 1624

fist thing i get hit with is package_version is too big as its got a +sha.(whatever) put on the end, so i edit that bit 

now im getting the timer0 problem 

so im just wondering if anyone could help me out with a bit more info on what i need to do to get the build button to work 

sorry for the noob question but very keen to try this and its just not working for me

TIA

MCUdude, 2022-02-10:
Use

```
platform = https://github.com/platformio/platform-atmelmegaavr#develop
```

instead

br74649, 2022-02-10:
Hi, 

because i hate to find problems without a resolution in google... i thought i would put my resolution here 

deleting everything with avr in it under .platformio/packages .platformio/platforms and deleting all of .cache made it suddenly work, when running the build it downloaded everything into a clean environment i suspect

doing as above in platform = https://github.com/platformio/platform-atmelmegaavr#develop

and all was well

My final issue was the test blink sketch used led_builtin which is not defined in the 1624 chip so was throwing a linker error in wiring.c, but led_builtin works for the 1626 which was a bit of a red herring for me to figure out

Thanks @MCUdude  for your help 

I have at least learned something - not sure what it is but i did learned it :)

closed 2022-02-10 by br74649