platformio/platform-atmelmegaavr #41: DXCore uart upload with optiboot

matou78, 2022-03-21:
Hello everyone

Found something that i think should be documented or fixed.

I started to play a little with the AVRxxDAxx chips using platformIO .
Using DXCore with arduinoIDE everything is working , but with plaformIO nothing worked (updi upload,bootloader burn,uart upload)

[maxgerhardt](https://community.platformio.org/u/maxgerhardt) on the platformIO forum helped me to find few problems,
This topic will be only about the UART upload with optiboot already on the chip

when i first started , i made a little program to tests the serial monitor and a blinking LED
I uploaded it through UART but nothing was happening
I tried with a DIY updi programmer based on CH340B, bought and tried mplab snap and pickit4 with same results

I took a look in logs from platformIO and arduinoIDE ( with verbose activated ), found some things that didn't match .
Found a flag named -DUSING_OPTIBOOT .

(As I'm a noob in this things, before i found a better solution ) 
I went inside of the .json board file and found same style of flags so added it there ... and voila, worked perfectly.

After few tries everything was working nicely, then google helped me a little and found that in platformio.ini, we can add something called "build_flags"

This is what looks like my settings to upload through UART, working only with optiboot already on the chip ( burned it with arduinoIDE)
```
[env]
platform = atmelmegaavr
board = AVR64DA32
framework = arduino
board_build.f_cpu = 24000000L

[env:Upload_UART]
build_flags = -DUSING_OPTIBOOT
upload_protocol = arduino
upload_speed = 115200
```
Is there a way to put it somewhere so people don't loose 2 weeks searching as i did ? or maybe put it directly in the config files .
Thanks

SpenceKonde, 2022-03-22:
Due to dubious decisions made by Microchip,. binaries m form optiboot and non-optiboot ae not compatible (thesectionstart business). It's recently been complicated by my realization that 99% of AVRs getting into a "bad state" tat requires power cycle or reset pin reset to recover from the cause is a "dirt reser"where execution ends up a 0x0000 without a reset having occurred, which is virtually guaranteed not to work. Since these are caused by user error, we can't prevent them, but it's the reason forthe two key behaviors involved in this:
1. Optiboot will ALWAYS clear the reset flags before entering the application. They are stored in GPIOR0. This is required both to detect dirty resets (if one is detected it immediately performs a software reset so that it has a prayer of functioning). 
2. If  USING_OPTIBOOT is set, we accept that this was already done. Otherwise, in .init3, we perform the same check. Bt if the bootloader as already cleared the flags, it  will look like a dirty reset so we'll issue a software reset and you'll bootloop. This requirement was added fairly recently

This is one of the very frustrating things about trying to develop with so many PIO users. I have no control over what defines are passed there. Yet that is a key part of my toolbox in arduino-land which I lean on very heavily -Tjere are cases where I require one ou of a number of defines  is part of my toolbox on Arduino, and I use it *very* heavily. I realize I need certain board definitions to have some define so I add it to platform.txt or use a regex to add t to the appropriate boards and then it just works, But it doesn;'t seem possible to make PIO do that sort of thing by default. and mot using it myself, I don't really even know wat to tell people to do.....

matou78, 2022-03-23:
i think that it was doing a bootloop like you said, sometimes the serial wasn't working at all and sometimes was spewing just a "t"
( i've put a Serial.println("test ok"); to see if serial was working as the led wasn't )

It is possible to do a default platformio.ini i think but sure, would love to see a little menu when creating the project to generate platformio.ini file with parameters set like in arduinoIDE .

Anyways, Thank you a lot for your work .

mikrocoder, 2022-08-20:
Thanks matou78. Finally I can flash via USB and bootloader. I have been missing these settings.

matou78, 2022-08-20:
happy that it helped :)