platformio/platform-atmelmegaavr #46: Upgrade to Avrdude 7.0

MCUdude, 2022-08-19:
[Avrdude](https://github.com/avrdudes/avrdude) 7.0 has been out for a while now, and Arduino has their own build with statically linked libraries that "just works".

There are lots of bug fixes and new features, but the most exciting one is probably support for SerialUPDI. It would be great if you could update Avrdude for "megaAVR's" as well as for "classic" AVRs.

Here are all the pre-compiled Arduino-build Avrdude binaries:
https://github.com/arduino/avrdude-packing

🇺🇦

matou78, 2022-08-22:
i updated it manually to gain access to serialupdi , didn't see any problems
Should be updated in my opinion

mairas, 2023-01-31:
Would be great to have this updated!

MCUdude, 2023-07-24:
Avrdude 7.1 is now default on MegaCoreX, megaTinyCore and DxCore

closed 2023-07-24 by MCUdude