platformio/platform-atmelmegaavr #49: ATMega4808 and bootloader with UPDI

JulesGren, 2022-11-07:
If one uses the Arduino IDE to load a bootloader (or hex including a bootloader) into a 4808 (perhaps also other chips as well) by default it places the bootload at0x0 and the main program at 0x200.

If you now try to load a program via platformio/vscode using UPDI, it loads this into memory at 0x0 and not 0x200, unfortunately the chip still looks for the start of the code at 0x200 therefore when trying to upload code from platformio it does not work (although it does upload successfully - it doesn't direct the chip to start in the correct place).

The workaround (a bit unreliable) is to upload blink or similar from Arduino IDE with the nobootloader option selected - a few times, at which point platformio can now be used to upload code via UPDI, it does mean platformio needs Arduino IDE to configure the chip first. 

Options:

1) Enable the user to select the start point as 0x0 or 0x200 - ideal as this retains the bootloader for serial upload and UPDI, or at 0x0 when there is no bootloader
2) Ensure that when uploading it configures the chip to start at 0x0 (noting that normal serial loads on a bootloader still need to start at 0x200)

brunob45, 2022-11-11:
Have you selected the correct bootloader option in your platformio.ini file?

You should have the line `board_hardware.uart=...`

Have a look at this for the explanation https://github.com/MCUdude/MegaCoreX/blob/master/PlatformIO.md

JulesGren, 2022-11-11:
Yes, have run it as blank (no ref to bootloader), and as no_bootloader and as uart1 / uart1_def, which is the bootloader installed.

There are slight differences, in some cases the in-built LED is illuminated in others it is not.

brunob45, 2022-11-19:
Could you try to include the flag `-DUSING_OPTIBOOT ` in the `build_flags`?

I made a pull request so it is included by default, but it is not yet merged.

JulesGren, 2022-11-21:
Doesn't appear to work.

The 4808 has a bootloader from Arduino installed, if I load with:
[env:ATmega4808]
platform = atmelmegaavr
board = ATmega4808
framework = arduino
;upload_protocol= arduino
;board_hardware.uart = uart1 / uart1_def
board_hardware.uart = no_bootloader
build_flags = -DUSING_OPTIBOOT

Then it builds and transfers ok, but doesn't flash (it's just std blink) this is the same if I set no bootloader and uart1.

build blink in arduino ide with no bootloader, then both no bootloader and uart1 bootloader may be loaded.

it appears as if platformio can't overwrite an arduino installed bootloader.

brunob45, 2022-11-21:
In platformio, installing the bootloader and the application is a 2 step process :
- the command `pio run - t bootloader` will install the bootloader
- the command `pio run - t upload` will install the application

matou78, 2022-11-25:
uploading with UPDI erases automatically the bootloader,
You need to reconfigure the chip flags by clicking the button "set flags" in PIO, this should help with your issue
I had similar issues with AVRxxDAxx chips
https://github.com/platformio/platform-atmelmegaavr/issues/42#issue-1175368640

JulesGren, 2022-11-29:
Nope, neither of these last two suggestions work.

Yes I can write the bootloader
Yes I can write the code

No I cannot overwrite an existing Arduino bootloader
No I cannot write code on an existing Arduino bootloader

All of the above works though if i don't use UPDI for writing code

brunob45, 2022-12-02:
The Arduino bootloader, as well as the Platformio bootloader (I think they are the same) enables the upload of a program using the serial port. 

If you are using UPDI, you will overwrite all of the memory, including the bootloader. 

When using UPDI with the Arduino IDE, it may re-upload the bootloader each time, so that's why it is working there but not Platformio.

Once the bootloader is installed, please do not use UPDI.

If you want to use UPDI, you do not need the bootloader.

matou78, 2023-02-03:
Arduino IDE selects which way you upload the program

Selecting " burn bootloader" the ide will use updi to put the bootloader
When pressing the button "upload" the ide automatically use serial if not specified other options

brunob45, 2023-02-12:
@matou78 correction :
> When pressing the button "upload" the ide automatically use serial if ~not specified other options~ the selected board is `AVR DA-series (Optiboot)`. The IDE uses UPDI if the selected board is `AVR DA-series (no bootloader) `.

paulskirk53, 2023-06-17:
I have the same issue trying to upload to ATmega4809 using jtag2updi. In arduino IDE 2, I have configured the IDE to use jtag2updi and it uploads and verifiies 15k ish of code in 6 seconds. The code runs.

If I configure Platformio using exactly the same hardware config, it uploads and verifies the same 15k ish code in 10 seconds - ( slower), but although avrdude reports success with 15k ish bytes verified, the code does not run.

Really looking forward to a fix :) , thanks.