platformio/platform-atmelmegaavr #53: Upgrade framework-arduino-megaavr-megatinycore to 2.6.4

mairas, 2023-01-31:
Upstream megaTinyCore has significant improvements and bugfixes. Could you please upgrade `framework-arduino-megaavr-megatinycore`?

Thanks! 😊

SpenceKonde, 2023-01-31:
@mairas 2.6.4? sometime in the early morning on monday 2.6.5 mTC along with 1.5.4 DxC was released 

Right now, to the best of my knowledge, both of the cores are in pretty good shape, and I don't forsee anything big in the next few months, so this would be a good time to do it,

mairas, 2023-02-01:
Cool! Didn't visit the page before opening this issue, so my info was a couple of days too old. :-D Updated the request title.

valeros, 2023-02-03:
Resolved in https://github.com/platformio/platform-atmelmegaavr/commit/89c1b5cbcdba7cbf5895a3c506a6822146780120

closed 2023-02-03 by valeros