platformio/platform-atmelmegaavr #55: framework-arduino-megaavr-dxcore outdated

fredrikvang, 2023-03-17:
Platform-atmelmegaavr currently depends an outdated version of dxcore (1.4.5 and 1.4.10 available on PlatformIO, but 1.5.6 available on GitHub and Arduino IDE). This causes build failures for AVR64DD28, and several bugs have been fixed in recent versions. 

See also post: https://community.platformio.org/t/workaround-for-avrxxddxx-dxcore-outdated/32610?u=mr.vang

brunob45, 2023-03-17:
Hi @fredrikvang 
I have forked the official PlatformIO package with the updated DxCore version.
You can find an example here: https://github.com/brunob45/dev-AVR-Dx

I'm using the custom platform https://github.com/brunob45/platform-atmelavrdx#dev/dxcore
and the custom framework https://github.com/brunob45/framework-arduino-megaavr-dxcore#platformio

I see that another user opened a similar issue with SpenceKonde/DxCore#319.
Usually, here is the correct place to request an update for the DxCore package

bgersmann, 2024-01-02:
Hi,

could you please update DxCore to  1.5.11 ?

Thanks