platformio/platform-atmelmegaavr #60: Pre-defined fuses ... can't set UPDI pin mode or MVIO mode fuses on AVR64DD28

tactical-snacks, 2023-05-30:
i think the pre-defined fuses for the AVR64DD28 leave the UPDI pin in gpio mode and set the MVIO register bit in a reserved configuration, neither yes or no. I could change the reset pin mode and bod level so that works.

perused [fuses.py](https://github.com/platformio/platform-atmelmegaavr/blob/af3bbedf54e5df6a22a8f9e7030030e7d31f7990/builder/fuses.py#L155) and didn't see an [updi pin](https://github.com/platformio/platform-atmelmegaavr/blob/af3bbedf54e5df6a22a8f9e7030030e7d31f7990/builder/fuses.py#L66) associated with the dxcore. Also, MVIO seems to be associated with [DB series only](https://github.com/platformio/platform-atmelmegaavr/blob/af3bbedf54e5df6a22a8f9e7030030e7d31f7990/builder/fuses.py#L155).

```
Selected fuses:
-------------------------
[fuse0 / wdtcfg   = 0x00]
[fuse1 / bodcfg   = 0x00]
[fuse2 / osccfg   = 0x00]
[fuse4 / tcd0cfg  = 0x00]
[fuse5 / syscfg0  = 0xC9]
[fuse6 / syscfg1  = 0x06]
[fuse7 / codesize = 0x00]
[fuse8 / bootsize = 0x00]
[lock  / lockbit  = 0x5CC5C55C]
-------------------------
```

threw these into microchip studio and compared them against the datasheet.

toolchain info ...
```
PLATFORM: Atmel megaAVR (1.9.0) > AVR64DD28
HARDWARE: AVR64DD28 24MHz, 8KB RAM, 64KB Flash
PACKAGES:
 - framework-arduino-megaavr-dxcore @ 1.5.6
 - tool-avrdude @ 1.70100.0 (7.1.0)
 - toolchain-atmelavr @ 3.70300.220127 (7.3.0)
```

valeros, 2023-05-30:
Hi @MCUdude, could you please take a look?

MCUdude, 2023-05-30:
I'm a bit busy at the moment, but I'll have a look when I have time. However, I don't recall adding DxCore support to the fuses.py script?

MCUdude, 2023-05-30:
I just looked at the commit history. I was me that added DxCore support to the fuses.py script 😄

MCUdude, 2023-05-30:
I haven't updated the fuses script to support AVR-DDs yet.