platformio/platform-atmelmegaavr #61: Atmelavr toolchain iom4808.h file missing USART3?

DAflamingFOX, 2023-06-03:
This is a copy of the discourse [thread](https://community.platformio.org/t/atmelavr-toolchain-iom4808-h-file-missing-usart3/34143) on the same topic, I was asked to make an issue here.

Hello, I would just like to preface this with: **I have no idea what I am doing and used my monkey brain pattern recognition to fix a bug**. That being said, I wanted to find some closure with the bug I found and fixed (I mean it at least programs now when it didn't before).

I'll try to be as detailed as possible, but again, not sure what I'm doing so I might miss something.

What I did was create a new platformio project by selecting atmega4808 as the board.
When this project was created I then copied the `jtag2updi` platformio.ini from the megacorex platformio docs [here](https://github.com/MCUdude/MegaCoreX/blob/master/PlatformIO.md).
After doing this, compiling the default sketch generated by platformio I got the following error:
```
In file included from C:\Users\ffejs\.platformio\packages\framework-arduino-megaavr-megacorex\cores\MegaCoreX\UART.h:28:0,
                 from C:\Users\ffejs\.platformio\packages\framework-arduino-megaavr-megacorex\cores\MegaCoreX\Arduino.h:127,
                 from C:\Users\ffejs\.platformio\packages\framework-arduino-megaavr-megacorex\cores\MegaCoreX\UART3.cpp:25:
C:\Users\ffejs\.platformio\packages\framework-arduino-megaavr-megacorex\variants\48pin-standard/pins_arduino.h:204:42: error: 'USART3' was not declared in this scope
 #define HWSERIAL3                       &USART3
                                          ^
```
There was a bit more but it was all USART3 stuff.
Control clicking through the files it listed (I'm not sure how I got to it as its not directly one listed) I got to the iom4808.h file which comes with the atmelavr toolchain. (the path is `C:\Users\ffejs\.platformio\packages\toolchain-atmelavr\avr\include\avr\iom4808.h`)
This file was missing:
* The memory mapping for USART3 
  * ```
    #define USART3              (*(USART_t *) 0x0860)
    ```
* The interrupt vectors for USART3
  * ```
    #define USART3_RXC_vect_num  37
    #define USART3_RXC_vect      _VECTOR(37)
    #define USART3_DRE_vect_num  38
    #define USART3_DRE_vect      _VECTOR(38)
    #define USART3_TXC_vect_num  39
    #define USART3_TXC_vect      _VECTOR(39)
    ```
After adding that to the file on my machine everything compiles and uploads fine.
(Yes, I checked the values with the datasheet and they are correct)

I've scoured the github repo's looking for this file and was not able to locate it, as I wanted to make a github issue about this topic, so instead I've resorted to this.

Is there any reason it was missing or was it just a mistake and a bug that should be fixed?
Or am I messing with something I'm not supposed to be messing with and there was a really simple solution to my problem.

DAflamingFOX, 2023-06-07:
I don't seem to be able to recreate the issue again either, not sure anymore lol

MCUdude, 2023-07-24:
The ATmega4808 doesn't have USART3, so maybe your program somehow was compiled for an ATmega480**9**, or a different 48-pin megaavr chip?