platformio/platform-atmelmegaavr #70: delay() not working on Atmega808

fatgorille, 2024-04-13:
I've encountered an issue with the delay() function on the ATMEGA808 microcontroller. When I run the following code, the uC freezes, and no signal is detected on the TEST_PIN:

`#include <Arduino.h>

#define TEST_PIN 24

void setup() {
  pinMode(TEST_PIN, OUTPUT);  // debug pin
}

void loop() {
  digitalWrite(TEST_PIN, HIGH);
  delay(10);
  digitalWrite(TEST_PIN, LOW);
  delay(10);
}`

However, when I replace 'delay(10)'; with 'delayMicroseconds(10000);', the code runs smoothly. Additionally, when I run this code on an ATMEGA4809, it works without any issues.

I noticed there's a similar issue reported here [https://github.com/MCUdude/MegaCoreX/issues/43], but I'm unsure if it applies to PlatformIO as well.

Could you please help me verify if this issue is specific to PlatformIO and provide guidance on how to address it?

fatgorille, 2024-04-14:
After closer check, the toolchain was not up to date.

The toolchain installed were 1.70300.191015 (7.3.0)

By specifying the latested toolchain in platformio.ini 3.70300.220127 it fixes my issue.

closed 2024-04-14 by fatgorille

RudyJM, 2025-01-10:
Thanks @fatgorille ISRs and millis() are now working for me!! Updating the toolchain is what did it. 

Here's my .ini for anyone else who comes accross this issue:
[env:ATmega808]
platform = atmelmegaavr
board = ATmega808
framework = arduino

; Clock and target configuration
board_build.f_cpu = 16000000L       ; Set clock speed to 16 MHz
board_build.mcu = atmega808         ; Specify the microcontroller

; Fuse settings
board_fuses.osccfg = 0x01           ; Internal 16 MHz oscillator
board_fuses.syscfg1 = 0x07          ; Maximum startup time
board_fuses.bodcfg = 0x54           ; BOD enabled at 2.6V
board_fuses.wdtcfg = 0x00           ; Watchdog Timer disabled
board_fuses.tcd0cfg = 0x00          ; Default timer configuration
board_fuses.syscfg0 = 0xC9          ; Reset pin, UPDI enabled
board_fuses.append = 0x00           ; Default application section end
board_fuses.bootend = 0x00          ; Default bootloader section end
board_fuses.lock = 0xC5             ; Default lock bits

; Update toolchain
platform_packages = toolchain-atmelavr@~3.70300.220127

; Custom upload protocol settings
upload_protocol = custom
upload_flags =
    -Pusb
    -cjtag3updi
upload_command = avrdude $UPLOAD_FLAGS -c jtag3updi -p $BOARD_MCU -U flash:w:$SOURCE:i