platformio/platform-atmelmegaavr #71: avrdude upload error when fuses are present in the elf file (AVR Dx)

kiiv-cz, 2024-08-18:
I'm getting error:

`avrdude error: address 0x820009 out of range at line 665 of .pio\build\test_fuses_in_elf\firmware.hex`

caused by having C source file containing FUSES setting (src/fuses.c):

```
#include <avr/io.h>

#ifdef MVIO_ENABLED
  #define MVIO_SETTING MVSYSCFG_DUAL_gc
#else
  #define MVIO_SETTING MVSYSCFG_SINGLE_gc
#endif

FUSES = {
    .WDTCFG   = WINDOW_OFF_gc | PERIOD_OFF_gc,
    .BODCFG   = LVL_BODLEVEL0_gc | SAMPFREQ_128Hz_gc | ACTIVE_DISABLE_gc | SLEEP_DISABLE_gc,
    .OSCCFG   = CLKSEL_OSCHF_gc,
    .SYSCFG0  = CRCSRC_NOCRC_gc | CRCSEL_CRC16_gc | RSTPINCFG_RST_gc | FUSE_EESAVE_bm,
    .SYSCFG1  = MVIO_SETTING | SUT_16MS_gc,
    .CODESIZE = 0x00,
    .BOOTSIZE = 0x00,
};
```

Fix is adding  "-R" ".fuse" into ElfToHex Builder [builder/main.py#L129](https://github.com/platformio/platform-atmelmegaavr/blob/develop/builder/main.py#L129):
  
```
        ElfToHex=Builder(
            action=env.VerboseAction(" ".join([
                "$OBJCOPY",
                "-O",
                "ihex",
                "-R",
                ".eeprom",
                "-R",
                ".fuse",
                "$SOURCES",
                "$TARGET"
            ]), "Building $TARGET"),
            suffix=".hex"
        )
```

Another workaround is using custom upload protocol and just copy hex file to CURIOSITY NANO drive:
```
upload_protocol = custom
upload_command = copy $SOURCE d:\
```

Board: Curiosity Nano DB / AVRxDB  (but should be valid for any supported MCU)
Plaform version: 1.9.0

brunob45, 2024-09-01:
Fuses are usually set by options in the platformio.ini file [as documented here](https://docs.platformio.org/en/latest/platforms/atmelmegaavr.html#fuses-programming). Setting the fuses manually elsewhere can have unforeseen consequences.

What fuse were you unable to set correctly?

kiiv-cz, 2024-09-03:
My aim is creating single production file (= elf) that contains code + eeprom and fuses so my coworker can just run Quick Programming tool from Microchip studio and not care about anything.

I can pass those fuses into program as defines to have it consistent (but only with magic numbers), however it still can't be present inside of hex (or I could use avrdude v8 that now supports flashing everything inside single file - released recently). 

Another way is another env with purpose only generate production file as changing platform localy will work only until it gets updated.