platformio/platform-atmelmegaavr #75: AVR-EA

uwezi, 2025-08-05:
I would like to use PlatformIO with the AVR-EA series of microcontrollers, using plain-C/C++ without the Arduino framework.
It appears to me that it should be easy to create a corresponding .json file and avr-gcc should already support the chips from this family. Also the DXCore (again, which I don't need) supports the AVR-EAs.

- But where would I need to place such a .json file?
- Or is it possible to just create an appropriate platformio.ini file with all the necessary settings?

valeros, 2025-08-05:
Hi @uwezi, take a look at this [page](https://docs.platformio.org/en/latest/platforms/creating_board.html).

uwezi, 2025-08-06:
> Hi [@uwezi](https://github.com/uwezi), take a look at this [page](https://docs.platformio.org/en/latest/platforms/creating_board.html).

thanks for the answer, I know that I yeas ago already got lost in the nested configuration scheme of platformio. At that point I wanted to create an environment for me which uses the plain ATmega328 with no Arduino or any other "framework". 

Now it's the same case, I don't want to use the dxcore or any other Arduino stuff or framework, just be able to program the AVR chip in plain C directly. Sadly there are neither examples nor documentation on how to achieve this. For the existing AVR64DA28 I found a solution by changing the `platformio.ini` of the project with the following settings, stripping out the `framework=` and applyingthe solution to issue #66:

```
[env:AVR64DA28]
platform_packages = platform/toolchain-atmelavr@^3.70300.220127
platform = atmelmegaavr
board = AVR64DA28
board_build.f_cpu = 4000000UL
upload_protocol = custom
upload_flags = -cserialupdi
  -p64da28
  -PCOM4
upload_command = "C:\Program Files (x86)\AVRDUDESS\avrdude.exe" $UPLOAD_FLAGS -U flash:w:$SOURCE:i
```

but when trying to modify a copy of the `AVR64DA28.json` into something like 
```
{
  "build": {
    "core": "dxcore",
    "f_cpu": "4000000L",
    "mcu": "avr64ea28",
    "variant": "28pin-standard"
  },
  "hardware": {
    "oscillator": "internal"
  },
  "frameworks": [
    "arduino"
  ],
  "name": "AVR64EA28",
  "upload": {
    "maximum_ram_size": 6144,
    "maximum_size": 65536,
    "protocol": "custom",
    "flags": "-cserialupdi -p64da28 -PCOM4",
    "command": "\"C:\\Program Files (x86)\\AVRDUDESS\\avrdude.exe\" $UPLOAD_FLAGS -U flash:w:$SOURCE:i"
  },
  "url": "https://www.microchip.com/en-us/product/avr64ea28",
  "vendor": "Microchip"
}
```
platformio complains
```
Resolving AVR64EA28 dependencies...
Tool Manager: Installing platform/toolchain-atmelavr @ ^3.70300.220127
UnknownPackageError: Could not find the package with 'platform/toolchain-atmelavr @ ^3.70300.220127' requirements for your system 'windows_amd64'
```

How does a completely stripped down version of a working `.json` file look like with just the essentials to invoke the correct version of avr-gcc?

uwezi, 2025-08-06:
```Tool Manager: Installing platform/toolchain-atmelavr @ ^3.70300.220127
UnknownPackageError: Could not find the package with 'platform/toolchain-atmelavr @ ^3.70300.220127' requirements for your system 'windows_amd64'```

but it is already there... the directory `C:\Users\uwezi\.platformio\packages\toolchain-atmelavr` contains this version, I also made a copy of the complete directory and named it `C:\Users\uwezi\.platformio\packages\toolchain-atmelavr@3.70300.220127` yet still PlatformIO always wants to revert to revert to `toolchain-atmelavr@1.70300.191015`

<img width="794" height="133" alt="Image" src="https://github.com/user-attachments/assets/d7137bdb-a952-46a7-8454-5c29160febb7" />

Please, I don't want to go back to Microchip Studio, nor to manual .make files...

valeros, 2025-08-06:
Your board manifest looks fine, the problem is here:
```ini
platform_packages = platform/toolchain-atmelavr@^3.70300.220127
```
It should be `platformio/toolchain-atmelavr` not `platform/toolchain-atmelavr`.
The bad news, however, is that the default toolchain doesn't support the AVR-EA series:
```
avr-gcc: error: device-specs/specs-avr64ea28: No such file or directory
```
As a workaround you can manually download a proper toolchain package from [DXCore](https://drazzy.com/package_drazzy.com_index.json) that supports the AVR-EA series and replace the default one in this folder `C:\Users\YOUR_USER\.platformio\packages\toolchain-atmelavr` just keep two files `.piopm` and `package.json`.

uwezi, 2025-08-06:
I found these update packages from Microchip - will try to get it running with these. Plan B is to stick with the AVR-DA for now and hope that Microchip at some point releases a complete toolchain...
https://packs.download.microchip.com/

I will update this topic here with my findings, as well as my homepage and git

uwezi, 2025-08-06:
Blink blink - success!

- I downloaded and unpacked the toolchain from https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino7b-i686-w64-mingw32.tar.bz2
- I then made a backup copy of the existing "default" toolchain under `.platformio/packages/toolchain-atmelavr` which contained the 3.7 toolchain
- I replaced the directories within the original  `.platformio/packages/toolchain-atmelavr`  with the files from the download
- I used the following board-definition .json:
```
{
  "build": {
    "f_cpu": "4000000L",
    "mcu": "avr64ea28",
    "variant": "28pin-standard"
  },
  "hardware": {
    "oscillator": "internal"
  },
  "name": "AVR64EA28",
  "upload": {
    "maximum_ram_size": 6144,
    "maximum_size": 65536,
    "protocol": "custom",
    "flags": "-cserialupdi -p64ea28 -PCOM4",
    "command": "\"C:\\Program Files (x86)\\AVRDUDESS\\avrdude.exe\" $UPLOAD_FLAGS -U flash:w:$SOURCE:i"
  },
  "url": "https://www.microchip.com/en-us/product/avr64ea28",
  "vendor": "Microchip"
}
```
- I then still got the error message 
```
[8/6/2025, 5:10:57 PM] Unable to resolve configuration with compilerPath "C:/Users/uwezi/.platformio/packages/toolchain-atmelavr@1.70300.191015/bin/avr-gcc.exe".  Using "cl.exe" instead.
```
since platformio still wanted to access the old toolchain, in spite of two newer ones installed in `toolchain-atmelavr` and `toolchain-atmelavr@3.70300.220127`
So I modified platformio.ini
```
[env:AVR64EA28]
platform_packages = platformio/toolchain-atmelavr
platform = atmelmegaavr
board = AVR64EA28
```
- this now let me write code with completion and highlighting, I also could compile the code, but the upload failed `avrdude: Can't find programmer id "custom"`
- it seems that giving the upload settings in the .json file always invokes the older avrdude provided by platformio, which does not support the `serialupdi` protocol. The `"command":` settings seems to be ignored.

So back to what worked before, changing the `platformio.ini` to 
```
[env:AVR64EA28]
platform_packages = platformio/toolchain-atmelavr
platform = atmelmegaavr
board = AVR64EA28

upload_protocol = custom
upload_flags = -cserialupdi
  -p64ea28
  -PCOM4
upload_command = "C:\Program Files (x86)\AVRDUDESS\avrdude.exe" $UPLOAD_FLAGS -U flash:w:$SOURCE:i
```
gave me my first blinking LED on an AVR64EA28.

Summary: it would be nice if the options and structure of the `board.json` files and the `platformio.ini` were documented somewhere in a clear manner. How do settings in the one file correspond to settings in the other?